raschy/ioBroker.deyeidc #19: fehler in deyeidc.MI600.json

Continueless, 2023-05-10:
deyeidc.MI600.json funktioniert auch für deye Sun800 das ist die gute Nachricht.

In der json ist aber ein fehler: DP1 = DC1*DV1 !!!

raschy, 2023-05-10:
Freut mich, dass der Adapter auch mit dem Sun800 läuft. Die 'deyeidc.MI600.json ist nur eine Beispieldatei, die nicht immer mitgepflegt wird. Sollte trotzdem richtig sein, wird beim nächsten Update berichtigt.

Continueless, 2023-05-10:
Nein, ich meine nicht, das der Adapter mit dem Sun800 funktioniert, denn das war ja klar. Sondern das die Register aus der Datei genau übereinstimmen mit den Registern des Sun800.

Continueless, 2023-05-10:
Doch ich benutze den Adapter, hatte nur so gut wie keine Arbeite, da die Register ja praktisch durch diese Datei mitgliefert wurden. Und das ist doch Perfekt. Finde nur, das im Adapter die ganzen Register schon eingetragen sein sollten, das die Leute, bei denen die Register noch nicht klar sind, diese nur Rausschmeissen, die nicht gebraucht werden und nicht umgekehrt.

raschy, 2023-05-10:
Da gehen unsere Meinungen etwas auseinander. Sollte ich einen Weg finden, verschiedene .json-Dateien einzuladen werde ich das einbauen. Bis dahin bleibt es dabei, dass der User eben keine Plug und Play Lösung nach deiner Vorstellung bekommt.

Continueless, 2023-05-10:
für mich ist das okay, bei mir läuft ja alles. Aber so werden nur sehr sehr wenige deine Arbeit tatsächlich nutzen. Ist halt schade um die Zeit meiner Meinung nach.

raschy, 2023-05-10:
Ich danke dir trotzdem für die Anregung und das Melden des Fehlers.

closed 2023-05-11 by raschy