raschy/ioBroker.deyeidc #35: Installation problem

lordvalium, 2023-08-05:
Hello, 
the only hint i found in the repo respective the readme.md is

`npm install iobroker.deyeidc`

My try:
http://192.168.188.46:8081/#tab-adapters --> searched for Deye. Does not appear. 

How can i attach my 12kW Deye to ioBroker?
Thanks

raschy, 2023-08-05:
The adapter is still not in the ioBroker repository, unfortunately. Therefore, an installation is also only possible via Github. According to our knowledge so far, the registers of the large inverters cannot be read. Or there is someone with newer knowledge (register directory).

lordvalium, 2023-08-06:
I will have a look if i find the registers

raschy, 2023-08-06:
Then I can only ask that you try the registers from Chapter 2 and report back here. Since I don't have a big inverter near me I can't test it.

closed 2023-08-09 by raschy