raschy/ioBroker.deyeidc #36: Adapter crash after update to 0.0.9

Reddingo, 2023-08-09:
**Describe the bug**  
When updating from 0.0.4 to 0.0.9 the adapter does not work anymore

**Screenshots & Logfiles**  
<html><body>

deyeidc.0 | 2023-08-09 10:46:09.716 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
-- | -- | -- | --
deyeidc.0 | 2023-08-09 10:46:09.715 | info | terminating
deyeidc.0 | 2023-08-09 10:46:09.710 | error | Cannot read properties of undefined (reading 'indexOf')
deyeidc.0 | 2023-08-09 10:46:09.709 | error | TypeError: Cannot read properties of undefined (reading 'indexOf')     at mathOperation (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:347:30)     at Deyeidc.readComputeAndWatch (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:316:22)     at Deyeidc.onData (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:170:16)
deyeidc.0 | 2023-08-09 10:46:09.708 | error | unhandled promise rejection: Cannot read properties of undefined (reading 'indexOf')
deyeidc.0 | 2023-08-09 10:46:09.707 | error | Unhandled  promise rejection. This error originated either by throwing inside of  an async function without a catch block, or by rejecting a promise which  was not handled with .catch().
deyeidc.0 | 2023-08-09 10:45:09.276 | info | starting.  Version 0.0.9 (non-npm: raschy/ioBroker.deyeidc) in  /opt/iobroker/node_modules/iobroker.deyeidc, node: v18.7.0,  js-controller: 4.0.24

</body>
</html>

**Versions:**  
 - Adapter version: 0.0.4 -> 0.0.9
 - JS-Controller version: 4.0.24
 - Node version: v6.3.5
 - Operating system: Ubuntu

raschy, 2023-08-09:
It is quite possible that when upgrading from 0.0.4 to 0.0.9, the structure in the configuration no longer fits. Therefore please delete the adapter completely and reinstall from Github.

Reddingo, 2023-08-09:
That fixed it. 
But I found another Bug, will report it in a different bugreport if you don't mind

closed 2023-08-09 by Reddingo

raschy, 2023-08-09:
I'm glad to hear that. I'll be happy to look at other bugs.