raschy/ioBroker.deyeidc #37: Reimport von JSON funktioniert nicht für Berechnete Werte

Reddingo, 2023-08-09:
Ich musste den Adapter neu installieren da das Updfate von 0.0.4 auf 0.0.9 nicht geklappt hat.
Ich habe vorher die Settings per JSON exportiert. 

Nach dem Reimport waren alle Werte wieder da bis auf die "Werte (Formeln)" für die 2 Berechnung von DP1 bzw DP2 bei mir (Key, Name, Einheit und Faktor waren korrekt importiert)

raschy, 2023-08-09:
Das ist auch zu erklären. In der vorigen Version gab es zwei Felder für die zu brechneneden Werte. Ab der Version 0.0.5 gibt es nur ein Feld "Wert**e**(Formeln)". Daher kann der Import der JSON aus der Version 0.0.4 nicht funktionieren. Daher bitte diesen Tab komplett neu ausfüllen (Ist ja hoffentlich nicht soviel).

Reddingo, 2023-08-09:
Okay. Das erklärt es. Waren zum Glück nur die 2 Power Berechnungen.

Und ich hab mich echt gefreut, dass du den Export und Reimport implementiert hattest, davor hat es mir echt gegraut die ganzen Werte neu einzugeben :-O

closed 2023-08-09 by Reddingo

raschy, 2023-08-09:
Ja gerne (ist aber bei den Adptern quasi im Standard schon drin). 
Da du den Sun600 hast, könntest du auch die Beispiel-JSON aus dem Repository (https://github.com/raschy/ioBroker.deyeidc/blob/main/deyeidc.MI600.json) nutzen. 
Dann musst du nur noch IP und Logger-SN eingeben.

Reddingo, 2023-08-09:
Fürs nächste mal weis ich bescheid, vielen Dank

Hast du eigentlich eine digitale Kaffeekasse?

raschy, 2023-08-09:
Nee lass mal. Ich mach das hier als Hobby - wenn ich helfen kan, gerne. Bis zum nächsten Bug

Reddingo, 2023-08-09:
Okay. Wie du magst :-) 
bis jetzt lief es ja stabil