raschy/ioBroker.deyeidc #39: ControlCode faulty!: Data may be corrupt, therefore discarded

OberfragGER, 2023-08-22:
I installed yesterday the 0.0.9 and everything is working fine.

This morning i recieve warnings:
`ControlCode faulty!: Data may be corrupt, therefore discarded`

I have set the start register to 0 and end to 90.

The adapter is still green and on debug but no further message but it seems to work fine. How to get rid of this warnings? Here an example of the whole logs.

[image]

`<html>
<body>

deyeidc.0 | 2023-08-22 11:11:49.951 | debug | connection closed/destroyed
-- | -- | -- | --
deyeidc.0 | 2023-08-22 11:11:29.951 | debug | Response: {"register":1,"modbus":{"type":"Buffer","data":[1,3,182,0,4,1,0,2,1,50,51,48,50,50,55,56,56,51,57,0,1,0,0,18,12,7,0,1,1,1,22,18,24,0,0,31,64,0,0,2,1,0,75,0,0,0,60,23,8,22,11,6,40,0,0,0,0,10,190,7,8,20,80,18,142,0,0,0,0,0,0,19,156,0,44,0,0,0,0,0,0,0,0,0,100,0,0,0,0,0,1,0,0,0,0,0,1,0,0,0,0,0,1,255,245,0,0,0,10,0,2,0,0,0,0,0,0,0,0,0,0,0,4,0,0,0,0,0,0,0,36,0,0,0,0,0,0,0,0,0,0,0,17,0,0,0,17,0,0,9,66,0,0,0,0,0,2,0,0,0,0,19,136,0,0,0,0,0,0,0,0,0,0,0,0,2,198,0,0,0,0,0,0,14,116,118]}}
deyeidc.0 | 2023-08-22 11:11:29.641 | debug | Response: {"register":0,"modbus":{"type":"Buffer","data":[1,3,2,22,11,247]}}
deyeidc.0 | 2023-08-22 11:11:29.579 | debug | connected
deyeidc.0 | 2023-08-22 11:11:29.542 | debug | try to connect . . .
deyeidc.0 | 2023-08-22 11:06:20.888 | debug | connection closed/destroyed
deyeidc.0 | 2023-08-22 11:06:00.888 | debug | Response: {"register":1,"modbus":{"type":"Buffer","data":[1,3,182,0,4,1,0,2,1,50,51,48,50,50,55,56,56,51,57,0,1,0,0,18,12,7,0,1,1,1,22,18,24,0,0,31,64,0,0,2,1,0,75,0,0,0,60,23,8,22,11,1,32,0,0,0,0,10,190,7,8,20,80,18,142,0,0,0,0,0,0,19,156,0,44,0,0,0,0,0,0,0,0,0,100,0,0,0,0,0,1,0,0,0,0,0,1,0,0,0,0,0,1,255,245,0,0,0,10,0,2,0,0,0,0,0,0,0,0,0,0,0,4,0,0,0,0,0,0,0,36,0,0,0,0,0,0,0,0,0,0,0,17,0,0,0,17,0,0,9,56,0,0,0,0,0,2,0,0,0,0,19,136,0,0,0,0,0,0,0,0,0,0,0,0,2,28,0,0,0,0,0,0,13,212,48]}}
deyeidc.0 | 2023-08-22 11:06:00.629 | debug | Response: {"register":0,"modbus":{"type":"Buffer","data":[1,3,2,22,11,247]}}
deyeidc.0 | 2023-08-22 11:06:00.565 | debug | connected
deyeidc.0 | 2023-08-22 11:05:59.542 | debug | try to connect . . .
deyeidc.0 | 2023-08-22 11:01:07.268 | debug | connection closed/destroyed
deyeidc.0 | 2023-08-22 11:00:47.268 | warn | ControlCode faulty!: Data may be corrupt, therefore discarded
deyeidc.0 | 2023-08-22 11:00:30.052 | debug | [watchStates] set to Power_Set
deyeidc.0 | 2023-08-22 11:00:29.989 | debug | Response: {"register":1,"modbus":{"type":"Buffer","data":[1,3,182,0,4,1,0,2,1,50,51,48,50,50,55,56,56,51,57,0,1,0,0,18,12,7,0,1,1,1,22,18,24,0,0,31,64,0,0,2,1,0,75,0,0,0,60,23,8,22,10,56,28,0,0,0,0,10,190,7,8,20,80,18,142,0,0,0,0,0,0,19,156,0,44,0,0,0,0,0,0,0,0,0,100,0,0,0,0,0,1,0,0,0,0,0,1,0,0,0,0,0,1,255,245,0,0,0,10,0,2,0,0,0,0,0,0,0,0,0,0,0,4,0,0,0,0,0,0,0,36,0,0,0,0,0,0,0,0,0,0,0,17,0,0,0,17,0,0,9,46,0,0,0,0,0,2,0,0,0,0,19,146,0,0,0,0,0,0,0,0,0,0,0,0,2,68,0,0,0,0,0,0,13,242,209]}}
deyeidc.0 | 2023-08-22 11:00:29.680 | debug | Response: {"register":0,"modbus":{"type":"Buffer","data":[1,3,2,22,10,54]}}
deyeidc.0 | 2023-08-22 11:00:29.617 | debug | connected
deyeidc.0 | 2023-08-22 11:00:29.543 | debug | try to connect . . .

</body>
</html>`

raschy, 2023-08-22:
How many of these warnings do you have? This control code is calculated in the Inverter and depends on ... no idea what. First ignore it and look again later. You can not change anything.

OberfragGER, 2023-08-22:
> How many of these warnings do you have? This control code is calculated in the Inverter and depends on ... no idea what. First ignore it and look again later. You can not change anything.

I think about every 30 or 60 Minutes. not very often.
I will update this post if i have better infos.

raschy, 2023-08-29:
No further info or explanation received within 7 days.

closed 2023-08-29 by raschy