raschy/ioBroker.deyeidc #46: Crashing after update to v0.0.11

Dabuna, 2023-09-04:
**Describe the bug**  
Chrashes after installation of 0.0.11

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
```
deyeidc.0 | 2023-09-04 09:30:57.957 | debug | checkUserData is ready
-- | -- | -- | --
deyeidc.0 | 2023-09-04 09:30:57.957 | debug | Sync time set to 60000 ms
deyeidc.0 | 2023-09-04 09:30:57.956 | debug | IP address [192.168.200.106] seems to be valid.
deyeidc.0 | 2023-09-04 09:30:57.943 | info | starting. Version 0.0.11 in /opt/iobroker/node_modules/iobroker.deyeidc, node: v18.17.1, js-controller: 4.0.24
deyeidc.0 | 2023-09-04 09:30:26.816 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
deyeidc.0 | 2023-09-04 09:30:26.816 | info | terminating
deyeidc.0 | 2023-09-04 09:30:26.815 | debug | [onUnload] cleaned everything up...
deyeidc.0 | 2023-09-04 09:30:26.813 | error | Cannot read properties of undefined (reading 'indexOf')
deyeidc.0 | 2023-09-04 09:30:26.813 | error | TypeError: Cannot read properties of undefined (reading 'indexOf')     at mathOperation (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:347:30)     at Deyeidc.readComputeAndWatch (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:316:22)     at Deyeidc.onData (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:170:16)
deyeidc.0 | 2023-09-04 09:30:26.813 | error | unhandled promise rejection: Cannot read properties of undefined (reading 'indexOf')
deyeidc.0 | 2023-09-04 09:30:26.812 | error | Unhandled  promise rejection. This error originated either by throwing inside of  an async function without a catch block, or by rejecting a promise which  was not handled with .catch().
deyeidc.0 | 2023-09-04 09:30:26.806 | debug | Response:   {"register":1,"modbus":{"type":"Buffer","data":[1,3,64,0,4,0,0,0,0,0,0,12,105,0,0,0,0,0,0,0,0,0,0,7,74,0,0,4,181,0,0,9,6,0,0,0,0,0,0,0,0,0,0,19,146,0,0,0,0,0,0,0,0,0,0,0,0,0,180,0,0,0,0,0,0,8,252,83]}}
deyeidc.0 | 2023-09-04 09:30:26.665 | debug | Response: {"register":0,"modbus":{"type":"Buffer","data":[1,3,2,4,9,122]}}
deyeidc.0 | 2023-09-04 09:30:26.603 | debug | connected
deyeidc.0 | 2023-09-04 09:30:26.593 | debug | try to connect . . .
deyeidc.0 | 2023-09-04 09:29:26.592 | debug | checkUserData is ready
deyeidc.0 | 2023-09-04 09:29:26.592 | debug | Sync time set to 60000 ms
deyeidc.0 | 2023-09-04 09:29:26.592 | debug | IP address [192.168.200.106] seems to be valid.
deyeidc.0 | 2023-09-04 09:29:26.579 | info | starting. Version 0.0.11 in /opt/iobroker/node_modules/iobroker.deyeidc, node: v18.17.1, js-controller: 4.0.24
deyeidc.0 | 2023-09-04 09:29:26.521 | debug | States connected to redis: 127.0.0.1:9000
deyeidc.0 | 2023-09-04 09:29:26.472 | debug | States create User PubSub Client
deyeidc.0 | 2023-09-04 09:29:26.472 | debug | States create System PubSub Client
deyeidc.0 | 2023-09-04 09:29:26.469 | debug | Redis States: Use Redis connection: 127.0.0.1:9000
deyeidc.0 | 2023-09-04 09:29:26.461 | debug | Objects connected to redis: 127.0.0.1:9001
deyeidc.0 | 2023-09-04 09:29:26.458 | debug | Objects client initialize lua scripts
deyeidc.0 | 2023-09-04 09:29:26.450 | debug | Objects create User PubSub Client
deyeidc.0 | 2023-09-04 09:29:26.450 | debug | Objects create System PubSub Client
deyeidc.0 | 2023-09-04 09:29:26.449 | debug | Objects client ready ... initialize now
deyeidc.0 | 2023-09-04 09:29:26.437 | debug | Redis Objects: Use Redis connection: 127.0.0.1:9001

```
**Versions:**  
 - Adapter version: v0.0.11
 - JS-Controller version: 4.0.24
 - Node version: v18.17.1
 - Operating system: Debian/Docker

**Additional context**  
Before I updated I think I had v0.0.4 installed and it didn't have any problems. The Deye is running and is reachable. It has firmware version 1.57

Dabuna, 2023-09-04:
Ok, for some reason the settings were lost. Reimported deyeidc.MI600.json and it works again.

closed 2023-09-04 by Dabuna

raschy, 2023-09-04:
This phenomenon is probably due to the fact that the formulas for the calculated values have changed.