raschy/ioBroker.deyeidc #47: "Error: Frame CheckSum faulty! D" nach Update des WRs auf MW3_16U_5406_2.06

MarkSau, 2023-09-08:
Hi,
ich habe gestern ein Update des WRs auf 2.06 gewagt. An sich tut alles soweit. Auch die Daten werden vom Adapter ausgelesen.
Jedoch erscheint minütlich die Meldung "Error: Frame CheckSum faulty! d" 

WR: Bosswerk BW-MI600 (ohne Zusatzrelais)
FW: MW3_16U_5406_2.06

Der Debug zeigt folgendes:

`<html>
<body>

023-09-08 08:40:10.309 - debug: deyeidc.0 (1188) Sync time set to 60000 ms
2023-09-08 08:40:10.310 - debug: deyeidc.0 (1188) checkUserData is ready
2023-09-08 08:41:10.312 - debug: deyeidc.0 (1188) try to connect . . .
2023-09-08 08:41:10.321 - debug: deyeidc.0 (1188) connected
2023-09-08 08:41:10.331 - error: deyeidc.0 (1188) Error: Frame CheckSum faulty! d
2023-09-08 08:41:10.386 - debug: deyeidc.0 (1188) Response: {"register":0,"modbus":{"type":"Buffer","data":[1,3,2,8,6,63]}}
2023-09-08 08:41:10.515 - debug: deyeidc.0 (1188) Response: {"register":2,"modbus":{"type":"Buffer","data":[1,3,64,0,4,0,0,0,0,0,0,26,34,0,0,0,0,0,0,0,0,0,0,13,121,0,0,11,202,0,0,9,76,0,0,0,0,0,0,0,0,0,0,19,136,0,0,0,0,0,0,0,0,0,0,0,0,3,72,0,0,0,0,0,0,11,154,179]}}
2023-09-08 08:41:10.688 - debug: deyeidc.0 (1188) Response: {"register":1,"modbus":{"type":"Buffer","data":[1,3,82,0,4,1,0,2,1,50,50,48,52,48,57,48,55,56,50,0,1,0,0,18,12,7,0,0,0,1,18,2,7,0,0,23,112,0,0,2,1,0,75,0,0,0,60,23,9,8,6,54,24,0,0,0,0,10,190,7,8,20,80,18,142,0,0,0,0,0,0,19,156,0,44,0,0,0,0,0,0,0,0,0,120,248]}}
2023-09-08 08:41:10.866 - debug: deyeidc.0 (1188) Response: {"register":2,"modbus":{"type":"Buffer","data":[1,3,64,0,4,0,0,0,0,0,0,26,34,0,0,0,0,0,0,0,0,0,0,13,121,0,0,11,202,0,0,9,76,0,0,0,0,0,0,0,0,0,0,19,136,0,0,0,0,0,0,0,0,0,0,0,0,3,72,0,0,0,0,0,0,11,154,179]}}
2023-09-08 08:41:10.948 - debug: deyeidc.0 (1188) Response: {"register":3,"modbus":{"type":"Buffer","data":[1,3,8,1,75,0,22,1,32,0,3,166]}}
2023-09-08 08:41:11.004 - debug: deyeidc.0 (1188) [watchStates] set to Power_Set
2023-09-08 08:41:11.005 - debug: deyeidc.0 (1188) [watchStates] set to DV1
2023-09-08 08:41:11.005 - debug: deyeidc.0 (1188) [watchStates] set to DC1
2023-09-08 08:41:11.005 - debug: deyeidc.0 (1188) [watchStates] set to DV2
2023-09-08 08:41:11.006 - debug: deyeidc.0 (1188) [watchStates] set to DC2
2023-09-08 08:41:11.058 - debug: deyeidc.0 (1188) Response: {"register":4,"modbus":{"type":"Buffer","data":[1,3,8,10,190,7,8,20,80,18,142,34]}}
2023-09-08 08:41:31.059 - debug: deyeidc.0 (1188) connection closed/destroyed
2023-09-08 08:42:10.311 - debug: deyeidc.0 (1188) try to connect . . .
2023-09-08 08:42:10.317 - debug: deyeidc.0 (1188) connected
2023-09-08 08:42:10.323 - error: deyeidc.0 (1188) Error: Frame CheckSum faulty! d
2023-09-08 08:42:13.047 - debug: deyeidc.0 (1188) Response: {"register":0,"modbus":{"type":"Buffer","data":[1,3,2,8,6,63]}}
2023-09-08 08:42:13.282 - debug: deyeidc.0 (1188) Response: {"register":1,"modbus":{"type":"Buffer","data":[1,3,82,0,4,1,0,2,1,50,50,48,52,48,57,48,55,56,50,0,1,0,0,18,12,7,0,0,0,1,18,2,7,0,0,23,112,0,0,2,1,0,75,0,0,0,60,23,9,8,6,54,24,0,0,0,0,10,190,7,8,20,80,18,142,0,0,0,0,0,0,19,156,0,44,0,0,0,0,0,0,0,0,0,120,248]}}
2023-09-08 08:42:16.628 - debug: deyeidc.0 (1188) Response: {"register":2,"modbus":{"type":"Buffer","data":[1,3,64,0,4,0,0,0,0,0,0,26,34,0,0,0,0,0,0,0,0,0,0,13,121,0,0,11,202,0,0,9,76,0,0,0,0,0,0,0,0,0,0,19,136,0,0,0,0,0,0,0,0,0,0,0,0,4,206,0,0,0,0,0,0,11,184,200]}}
2023-09-08 08:42:19.893 - debug: deyeidc.0 (1188) Response: {"register":3,"modbus":{"type":"Buffer","data":[1,3,8,1,78,0,33,1,48,0,3,7]}}
2023-09-08 08:42:21.192 - debug: deyeidc.0 (1188) Response: {"register":4,"modbus":{"type":"Buffer","data":[1,3,8,10,190,7,8,20,80,18,142,34]}}
2023-09-08 08:42:41.192 - debug: deyeidc.0 (1188) connection closed/destroyed

</body>
</html>`

raschy, 2023-09-08:
Da gibt es offensichtlich ein Problem in der neuen Firmware. Da ich noch die 'alte' Firmware 1.57 habe, kann ich den Fehler nicht näher eingrenzen. Daher bitte mal die modifizierte Version 0.0.11 direkt von Github laden.

RP70DP, 2023-09-19:
Leider auch mit 0.0.11 nur noch Meldungen im Sekundentakt (aktuelle Firmware MW3_16U_5406_2.32-D1):

`2023-09-19 08:32:50.143  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! d
2023-09-19 08:32:51.334  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:32:51.934  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:32:52.535  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:32:53.134  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:32:53.740  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 58
2023-09-19 08:32:54.934  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:32:55.534  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:32:56.135  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:32:56.734  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:32:57.336  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 59
2023-09-19 08:32:58.538  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:32:59.132  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:32:59.732  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:00.333  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:02.134  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:33:02.734  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:03.334  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:03.936  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:04.539  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 58
2023-09-19 08:33:05.734  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:33:06.336  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:06.934  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:07.534  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:08.138  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 59
2023-09-19 08:33:09.334  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:33:09.934  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:10.534  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:11.134  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:12.935  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:33:13.534  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:14.134  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:14.734  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:15.338  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 58
2023-09-19 08:33:16.534  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:33:17.134  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:17.735  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:18.334  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:18.938  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 59
2023-09-19 08:33:20.141  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:33:20.734  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:21.334  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:21.935  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:23.734  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:33:24.339  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:24.934  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:25.534  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:26.138  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 58
2023-09-19 08:33:27.337  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:33:27.935  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:28.540  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:29.134  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:29.739  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 59
2023-09-19 08:33:30.934  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:33:31.535  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:32.134  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 53
2023-09-19 08:33:32.737  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:33:34.536  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
[... cut ...]
2023-09-19 08:36:54.342  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
2023-09-19 08:36:54.944  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 59
2023-09-19 08:36:56.141  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 7a
2023-09-19 08:36:56.743  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! be
2023-09-19 08:36:57.340  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! be
2023-09-19 08:36:57.942  - [31merror[39m: deyeidc.0 (811) Error: Frame CheckSum faulty! 1
`

raschy, 2023-09-19:
Die Firmware _2.32-D1 ist offensichtlich nicht kompatibel mit der Firmware 1.57. Da ich noch noch kein Relais habe und somit auch nicht die neueste Version bleibt einmal nur **direkt** von Github installieren.

RP70DP, 2023-09-19:
Ich nutze ja bereits 0.0.11.

raschy, 2023-09-19:
Hast du die Version von Github direkt installiert. Die hat die gleiche Versionsnummer wie die Installation per Admin. Ich kann derzeit keine Fehlerbehebung im klassichen Sinne machen.

RP70DP, 2023-09-19:
Ah so, weiß ich nicht mehr. Ich installiere mal von Github drüber!

RP70DP, 2023-09-19:
Das Frame CheckSum Problem scheint damit gelöst und es kommen auch wieder Daten rein. Aber jetzt wird das Log mit folgender Meldung überflutet:

`2023-09-19 11:42:25.963  - [32minfo[39m: host.iobroker instance system.adapter.deyeidc.0 started with pid 6220
2023-09-19 11:47:58.341  - [31merror[39m: deyeidc.0 (6220) TypeError: Cannot read properties of undefined (reading 'registerStart')
2023-09-19 11:47:58.608  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:47:59.210  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:47:59.813  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 10. Received 11
2023-09-19 11:48:00.410  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 7. Received 11
2023-09-19 11:48:02.207  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:02.807  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:03.412  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 10. Received 11
2023-09-19 11:48:04.011  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 7. Received 11
2023-09-19 11:48:05.814  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:06.406  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:07.007  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 10. Received 11
2023-09-19 11:48:09.408  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:10.007  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:10.607  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 10. Received 11
2023-09-19 11:48:11.213  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 7. Received 11
2023-09-19 11:48:13.007  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:13.609  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:14.206  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 10. Received 11
2023-09-19 11:48:14.810  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 7. Received 11
2023-09-19 11:48:16.607  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:17.208  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:17.808  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 10. Received 11
2023-09-19 11:48:20.206  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:20.808  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:22.443  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 10. Received 11
2023-09-19 11:48:25.612  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 7. Received 11
2023-09-19 11:48:27.407  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:28.006  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:28.607  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 10. Received 11
2023-09-19 11:48:31.007  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
[... cut ...]
2023-09-19 11:48:38.810  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 13
2023-09-19 11:48:39.415  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 10. Received 11
2023-09-19 11:48:43.614  - [31merror[39m: deyeidc.0 (6220) RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 7. Received 11`

raschy, 2023-09-19:
Gelöst ist das Problem damit noch nicht - es gibt nur einfach keine Warnung mehr. Ob die Daten sauber sind, kann ich nicht beurteilen. Möglicherweise gibt es bei irgendeinem Wert dann einfach Quatsch, den ich mit dem Fehler abgefangen hätte.

Zu dem aktuellen Log kann ich nichts sagen. Bei mir - alte Firmware - läuft der Adapter seit Tagen ohne diesen Fehler. 
Evtl. nochmal durch die Settings gehen und schauen, ob da alles ok ist.

tgs64, 2023-09-29:
Exakt das selbe hier bei mir: nach Einbau des Relais erst das Checksum-Problem, dann nach Installation des Adapters direkt von hier das "offset"-Problem. Settings durchgegangen -> nix gefunden.

Völlig unabhängig von diesem aktuellen Problem sei aber mal gesagt: tausend Dank für diesen Adapter und für deine Arbeit!

raschy, 2023-09-29:
Danke für den Hinweis. Zur zwischenzeitlichen Lösung siehe Post #4 hier oben.

tgs64, 2023-09-29:
Danke für die prompte Antwort!

Das ist aber leider keine "zwischenzeitliche Lösung": nach der Installation direkt von Github ändern sich (genau wie bei RP70DP) ja nur die Fehlermeldungen von "checksum" in "offset" ...

raschy, 2023-09-29:
Das tut mir sehr leid. Aber ohne Relais und somit ohne die neue Firmware habe ich keine Chance, den Fehler vernünftig zu fixen.
Relais ist bestellt - dauert.

tgs64, 2023-09-29:
Um Gottes Willen! Das ist doch kein Ding. Kein Stress!

Ich finde das megaklasse, dass du sowas machst, der Allgemeinheit zur Verfügung stellst und dann auch noch sofort bei Problemen antwortest! Danke!!!

raschy, 2023-10-09:
Die 'RangeErrors' stammen von der Kommunikation mit dem Relais. Zunächst mal herausgefiltert und wieder lauffähig gemacht ==> 0.0.12.

closed 2023-10-09 by raschy

ak68-hub, 2025-03-08:
Gibt es schon eine Lösung für das CheckSum Problem, da dieses offensichtlich  auch bei mir auftritt:

[image]

raschy, 2025-03-08:
Wie oft kommen denn die Meldungen zu "Frame Checksum faulty"? Häufig oder speziell nur beim Start des Wechselrichters?

ak68-hub, 2025-03-08:
Leider IMMER, es erfolgt kein Datenempfang:

[image]

raschy, 2025-03-08:
Sieht nach einem größeren Problem aus, Frage: Welcher WR ist das und hat es schon mal (in einer früheren Version) funktioniert?
Bitte mal die Logstufe des Adapter auf "Alles" stellen und das Log hier hinein stellen.

ak68-hub, 2025-03-08:
danke für Deine Hilfe :)

Log ist schon auf "Alles" und Protokoll 4 Posts weiter oben :)
WR: Deye SUN 12K, die Abfrage mit deyeidc hat noch NICHT funktioniert !

raschy, 2025-03-08:
Bei Log auf "Alles" müssten neben "debug" auch "silly" drin sein. Gerade wenn ich den CheckSum-Fehler finden soll.
Der Port 8899 sollte standardmäßig offen sein - es sei denn, Deye hätte sie durch ein Update geschlossen. Der Port reagiert wahrscheinlich nicht auf Telnet.
Die Weboberfläche des WR ist ein anderes Thema.

ak68-hub, 2025-03-08:
[image]

raschy, 2025-03-09:
Sorry, ich hatte meine Logfunktion deaktiviert. Kannst du bitte direkt von Github die aktuelle Version (Versionsnummer ist die gleiche) installieren? Und dann nochmal log mit "Alles".

ak68-hub, 2025-03-09:
Neues Log:

[image]

raschy, 2025-03-09:
Danke für das Log. Bitte nochmal von Github installieren, ich habe die Prüfung vorübergehend deaktiviert. 
Dann bitte mal schauen, ob Daten kommen und die plausibel sind.

ak68-hub, 2025-03-09:
Leider kommen wieder keine Daten :(

[image]

Im Minutenabstand folgend:

[image]

PS: Ich verwende einen SUN-12K-SG04LP3 mit folgendem JSON-Abrufdaten:

[Sun-12k-sg04lp3-eu.json]

raschy, 2025-03-09:
Bitte nochmal von Github installieren, ich habe eine weitere Prüfung vorübergehend deaktiviert. Bitte testen.

ak68-hub, 2025-03-09:
Ich weiss nicht, ob das gut oder schlecht ist, zumindest kommt jetzt eine Fehlermeldung:

[image]

raschy, 2025-03-09:
Richtig gut ist das noch nicht. Evtl. müssen wir jetzt mal klein Anfangen: Nicht alle Register und Spulen eintragen sondern nur einen (kleinen) Registerbereich und nur ein oder zwei Spulen. Irgendwo verschluckt sich der Adapter.

ak68-hub, 2025-03-10:
Jetzt habe ich nur die Register 587 und 588 abgefragt, leider gleiches Problem:

[image]

raschy, 2025-03-10:
Dann bin ich mit meinem Latein auch am Ende. Die Antworten vom WR, die ich erwarten würde (siehe oben) kommen ja nicht. Entweder kommt da auch nichts oder da antwortet jemand anderes.

ak68-hub, 2025-03-11:
> Dann bin ich mit meinem Latein auch am Ende. Die Antworten vom WR, die ich erwarten würde (siehe oben) kommen ja nicht. Entweder kommt da auch nichts oder da antwortet jemand anderes.

schade, .....
trotzdem vielen Dank für Deine Hilfe, vielleicht fällt Dir ja noch ´was ein :)