raschy/ioBroker.deyeidc #55: Adapter Objects didnt get autocreated

raidsec, 2023-10-19:
Hello,

i created the required register and config entries for my deye inverter:

[image]

[image]

But after multiple restarts the Adapter Objects are not getting autocreated. Is this normal?

[image]

[image]

Do i need to create the entries manually?

raschy, 2023-10-19:
No, the individual data points are created by the adapter. It may help to stop and restart the adapter.

raidsec, 2023-10-19:
Sure, have done a restart it multiple times. Also already reinstalled the adpter.
 Objects still did not get created.

raschy, 2023-10-19:
I just reinstalled the adapter again. Here it has created all data points. Maybe it helps to stop the Adpter, delete the data tree 'deyeidc.0' and then start it again.

raschy, 2023-10-19:
Node.js and npm are up to date and ok. Have you deleted the object tree?

raidsec, 2023-10-21:
I created a new IOBroker Installation and tried your adapter again. It is working so it looks like a problem with my local installation. Your Adapter is fine. Thank you for your work.

closed 2023-10-21 by raidsec