raschy/ioBroker.deyeidc #73: Register schreiben möglich?

octane321, 2024-01-17:
Hi raschy,
erstmal vielen Dank für den Adapter - habe eben meinen MI600 konfiguriert. Sobald die Sonne aufgeht sollten da Werte reinkommen.

Ich habe noch nen SUN-3.6K-SG03LP1-EU welchen ich momentan über RS485 via Modbus-Adapter auslese und beschreibe - das funktioniert auch.
Würde aber gerne auf den TCP Server vom Logger gehen statt des RS485 Adapters.
Auslesen funktioniert erstmal - aber können über den deyeidc-Adapter auch Register geschrieben werden?

Falls du auch Erfahrung hast, wie ich mit dem Modbus-Adapter auf den Logger connecten kann, wäre ich dir auch sehr dankbar.
VG

raschy, 2024-01-22:
Mit dem Modbus-Adapter komme ich auch nicht zurecht. Darum habe ich ja diesen Adapter entwickelt! Welche Register möchtest du denn beschreiben?

octane321, 2024-01-22:
Ich hab das komplette Modbus Protokoll durchgeackert und in den Modbus Adapter eingetragen.
Also grundsetzlich möchte ich den komplett via RS485 steuern und alle einstellungen machen.
Mit dem Modbus Adapter komme ich sehr gut zurecht - ich kann alles schreiben und lesen was es im Inverter gibt und sogar noch mehr, was im Menü garnicht zu erreichen ist. Aber nur mit einem USB-RS485 Adapter, der an den Inverter angeschlossen ist.
Versuche ich das ganze über den Modbus TCP Server der auf dem Logger läuft, bekomme ich ihn nicht verbunden.
Gibt es da irgend einen Trick? Wie läuft die Authentifizierung ab? Muss die Seriennummer dafür genutzt werden?

raschy, 2024-01-22:
Nach meinem Verständnis sprichst du über den Weg mit dem USB-RS485 direkt mit dem Inverter, über Modbus TCP nur mit dem Logger, der dem Inverter wohl vorgeschaltet ist. Demnach denke ich, wird das dann so nicht funktionieren.

octane321, 2024-01-22:
Verstehe leider nicht was du damit sagen willst.

Sagen wir mal ich möchte Register 245 "max Solar Sell (W)" schreiben, ist das über deinen Adapter möglich?
Die Schnittstelle des Loggers selbst dürfte nicht das Problem sein, denn über die Cloud kann man auch Einstellungen an den Inverter schicken.

Ich verbinde mich per TCP auf dem logger Port 8899 - die Verbindung steht.
Nun frage ich das Register 245 von Modbus ID 1 ab, bekomme aber keine Antwort. 

Kannst du mir verraten, mit welchen Parametern sich dein Adapter verbinden kann und welche Modbus ID der Logger hat?

raschy, 2024-07-01:
No new findings. Therefore closed from my side.

closed 2024-07-01 by raschy