raschy/ioBroker.deyeidc #74: RangeError

Harper-Chili, 2024-01-22:
Hi raschy,

ich habe einen DEYE SUN 12K-SG04LP3-EU. Den WR lese ich momentan über den Modbus-Adapter aus. Funktioniert auch soweit.
Jetzt bin ich auf deinen Adapter gestossen und probiere diesen zum laufen zu bringen.
Die Verbindung wird aufgebaut, doch kann ich keine Daten auslesen. Es kommt immer die untenstehende Fehlermeldnung.

Ich habe den Register Range von 500-520 festgelegt und die folgenden Daten eingeben. Die Daten habe ich deiner Definintion vom SUN10K entnommen. Diese entsprechen auch dem SUN12K.

"register": 514,
"rules": 1,
"key": "Battery_Today_Charge",
"name": "Battery_Today_Charge",
"name": "Battery Today Charge",
"unit": "kWh",
"factor": 1,
"nullable": false

"register": 515,
"rules": 1,
"key": "Battery_Today_Discharge",
"name": "Battery_Today_Discharge",
"name": "Battery Today Discharge",
"unit": "kWh",
"factor": 1,
"nullable": false

[requestData]  error: RangeError [ERR_OUT_OF_RANGE]: The value of "value" is out of  range. It must be >= 0 and <= 4294967295. Received 27_964_892_196  stack: RangeError [ERR_OUT_OF_RANGE]: The value of "value" is out of  range. It must be >= 0 and <= 4294967295. Received 27_964_892_196     at new NodeError (node:internal/errors:405:5)     at checkInt (node:internal/buffer:74:11)     at writeU_Int32LE (node:internal/buffer:694:3)     at Buffer.writeUInt32LE (node:internal/buffer:707:10)     at idcCore.requestFrame  (/opt/iobroker/node_modules/iobroker.deyeidc/lib/idc-core.js:275:10)     at Deyeidc.requestData  (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:225:29)     at async /opt/iobroker/node_modules/iobroker.deyeidc/main.js:92:5

Wo mache ich einen Fehler?

Gruß
Ralph

foxriver76, 2025-03-12:
Danke schon mal für deine Rückmeldung, ich helfe leider nur jemanden aus und habe das Gerät selbst nicht. Derjenige hat allerdings auch die Seriennummer in Verdacht und ist sich nicht sicher ob "die auf dem W-LAN /LAN Adapter oder die aus dem Gerät" gemeint ist - weißt du dazu etwas? ;-)

raschy, 2025-03-12:
Es muss die "Logger-Seriennummer" sein. Viele Kollegen haben den Adapter auch mit den großen WR ohne Fehler im Einsatz. (Sonst gäbe es wahrscheinlich mehr Issues).

foxriver76, 2025-03-15:
Hm okay, der Kollege sagt "LSW-3 damit klappt es, ich habe zwar nicht die Werte einzeln überprüft. Aber es erfolgt definitiv eine Ausgabe. Der Logger LSE-3 wird nicht erkannt". 

Hast du da evtl. eine Idee, warum der LAN Adapter nicht gehen sollte? Irgendwelche Logausschnitte die helfen könnten das einzugrenzen? 

Schönes WE.

raschy, 2025-03-15:
Ich muss mich erstmal schlau machen, was "LSW-3" bzw. "LSE-3" genau sind. Hat der Kollege denn die Möglichkeit dort irgendwo einen RS485/Modbus-Wandler anzuschließen und dann mal mit wenigen Registern über den Modbus Adapter auszulesen?
 
BTW: Um welchen Fehler geht es eigentlich hier genau? Um den Range-Error oder generelles Kommunikationsproblem?

raschy, 2025-03-24:
Hi, danke für die Info. Es scheint ja tatsächlich WR zu geben, die mit dem LSW arbeiten. 
Da müssen wir nochmal bei Adam und Eva anfangen: Welche SN ist eingetragen? Du siehst ja in den Logs, dass eine gültige Anfrage an den WR (Request to register set 1 >
a5170010450b007777d07d020000000000000000000000000000010301f40016840a5115) gesendet wird. Wenn die verwendete SN nicht passt, gibt es keine Antwort. Schau doch da bitte nochmal nach.

raschy, 2025-03-27:
Können wir uns darauf einigen, dass das Auslesen über den **LSE** nicht möglich ist! Hier wird ein anderes Sprachmodell verwendet, dass mit den im Adapter verwendeten Methoden nicht kompatibel ist. Der Aufwand dies zu Sondieren ist leider von mir nicht leistbar. Offensichtlich können die Daten - wie schon öfter angesprochen - direkt mit dem Modbus-Adapter ausgelesen werden.