raschy/ioBroker.deyeidc #75: Deye Sund 300G3-EU Register

Eelisglitchy, 2024-01-28:
Hi,

ich bin ziemlicher Neuling mit diesem Adapter.

Ich habe einen oben genannten WR finde aber keine Konfig Vorlage.

Ich habe hier : [Deye300G3-EU](https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_micro.md) wohl die Modbus Daten gefunden.

Kann mir jemand helfen die hier zu konfigurieren?

Danke sehr und LG.

raschy, 2024-01-28:
Für den Deye300 habe ich aktuell keine Config, Vielleicht nimmst du die vom 600er (https://github.com/raschy/ioBroker.deyeidc/blob/main/deyeidc.MI600.json) und löschst die nach deiner Konfig-Liste die nicht benötigten Coils.

Eelisglitchy, 2024-01-28:
Wow du bist aber wirklich schnell.

Danke für die schnelle Antwort, wusste nicht dass die config vom 600er ähnlich ist, werd mich mal ransetzen.

raschy, 2024-01-28:
Der hat halt einiges doppelt!

Eelisglitchy, 2024-01-29:
Guten Tag,

also ich hab gestern alles folgendermaßen erstellt.

Ich hab [von Hier](https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_micro.md) die Modbusliste genommen und die Coils vom Sun600 laut Vorlage hier. Hab immer gegenverglichen was der 300 wohl kann und bin nur von PV1 ausgegangen allso 1 Modul am WR.

Heute ist das Modul mit der Sonne wieder online gegangen, ich messe ja auch per Tasmota device noch die Leistung usw.

Jedenfalls der Deye Adapter ist grün alles scheint soweit i.O. Aber ich kriege einfach keine Daten.

[image]

[image]

Hier noch 2 Ausschnitte der Configs.
Im Protokoll alles normal sagt er mir dass er gestartet ist und alle 60 sek abfragt.
.

raschy, 2024-01-29:
Kannst du noch die 'Register' zeigen.

Eelisglitchy, 2024-01-29:
[image]

Ich muss auch dazu sagen, ich frage die Daten per Wiregaurd VPN von Fritze zu Fritze ab. Allso vom Schrebergarten nach Hause. Die Weboberfläche des WR ist dabei erreichbar. Und auch der Adapter sagt ja Verbindung grün.

raschy, 2024-01-29:
Ja, sieht soweit alles korrekt aus. Schonmal den Adpter gestoppt un neu gestartet?

Eelisglitchy, 2024-01-29:
Das habe ich auch mehrfach sogar. Auch mein Iobroker startet sich Sonntags neu, bzw der komplette Server. Da läuft auch noch ein Pihole, aber da alles lokal läuft dürfte der nicht dazwischenfunkten.

raschy, 2024-01-29:
Dann  bleibt noch die Möglichkeit, in der Instanzeinstellung mal auf 'debug' setzen und dann mal ins Protokoll schauen.

Eelisglitchy, 2024-01-29:
Interessant obwohl grün, scheint irgendwas abzureisen. Liegt es vllt am Port?

2024-01-29 12:03:40.255 - info: host.eelisglitchy-server "system.adapter.deyeidc.0" enabled
2024-01-29 12:03:40.496 - info: host.eelisglitchy-server instance system.adapter.deyeidc.0 started with pid 2001641
2024-01-29 12:03:40.906 - debug: deyeidc.0 (2001641) Redis Objects: Use Redis connection: 0.0.0.0:9001
2024-01-29 12:03:40.924 - debug: deyeidc.0 (2001641) Objects client ready ... initialize now
2024-01-29 12:03:40.925 - debug: deyeidc.0 (2001641) Objects create System PubSub Client
2024-01-29 12:03:40.926 - debug: deyeidc.0 (2001641) Objects create User PubSub Client
2024-01-29 12:03:40.948 - debug: deyeidc.0 (2001641) Objects client initialize lua scripts
2024-01-29 12:03:40.951 - debug: deyeidc.0 (2001641) Objects connected to redis: 0.0.0.0:9001
2024-01-29 12:03:40.970 - debug: deyeidc.0 (2001641) Redis States: Use Redis connection: 0.0.0.0:9000
2024-01-29 12:03:40.977 - debug: deyeidc.0 (2001641) States create System PubSub Client
2024-01-29 12:03:40.978 - debug: deyeidc.0 (2001641) States create User PubSub Client
2024-01-29 12:03:41.035 - debug: deyeidc.0 (2001641) States connected to redis: 0.0.0.0:9000
2024-01-29 12:03:41.157 - info: sql.0 (3405935) enabled logging of deyeidc.0.info.connection, Alias=false, WriteNulls=false
2024-01-29 12:03:41.166 - info: deyeidc.0 (2001641) starting. Version 0.0.14 in /opt/iobroker/node_modules/iobroker.deyeidc, node: v18.19.0, js-controller: 5.0.17
2024-01-29 12:03:41.174 - debug: deyeidc.0 (2001641) IP address [192.168.21.29] seems to be valid.
2024-01-29 12:03:41.175 - info: deyeidc.0 (2001641) Retrieving data from the inverter will be done every 60 seconds
2024-01-29 12:03:41.175 - debug: deyeidc.0 (2001641) checkUserData is ready
2024-01-29 12:03:41.175 - debug: deyeidc.0 (2001641) try to connect . . .
2024-01-29 12:03:41.288 - debug: deyeidc.0 (2001641) Connected to server
2024-01-29 12:03:51.339 - debug: deyeidc.0 (2001641) Connection timeout
2024-01-29 12:03:51.340 - debug: deyeidc.0 (2001641) Connection closed/destroyed
2024-01-29 12:03:51.342 - debug: deyeidc.0 (2001641) Connection closed
2024-01-29 12:04:07.977 - info: connector-local-control.0 (3407453) receive server message from 192.168.20.62: 32100: {"msgType":"Heartbeat","mac":"8caab5ca9f50","deviceType":"02000001","msgID":"20240129120404896","token":"967DC7CD55376DA9","data":{"currentState":1,"numberOfDevices":1,"RSSI":-41}}
2024-01-29 12:04:07.978 - info: connector-local-control.0 (3407453) Heartbeat

raschy, 2024-01-29:
Das sollte aber etwa so aussehen:

`<html>
<body>

2024-01-29 12:05:08.002 - debug: deyeidc.0 (179666) try to connect . . .
--
2024-01-29 12:05:08.182 - debug: deyeidc.0 (179666) Connected to server
2024-01-29 12:05:08.384 - debug: deyeidc.0 (179666) Response: {"register":1,"modbus":{"type":"Buffer","data":[1,3,82,0,4,1,0,2,1,50,50,49,48,48,53,48,57,50,55,0,0,0,0,18,12,7,0,1,2,1,39,18,21,0,0,78,32,0,0,4,1,0,75,0,0,0,60,24,1,29,12,4,58,0,0,0,0,10,190,7,8,20,80,18,142,0,0,0,0,0,0,19,156,0,44,0,0,0,0,0,0,0,0,0,75,159]}}
2024-01-29 12:05:08.571 - debug: deyeidc.0 (179666) Response: {"register":2,"modbus":{"type":"Buffer","data":[1,3,8,0,0,10,190,7,8,20,80,50]}}
2024-01-29 12:05:19.003 - debug: deyeidc.0 (179666) Connection timeout
2024-01-29 12:05:19.004 - debug: deyeidc.0 (179666) Connection closed/destroyed
2024-01-29 12:05:19.004 - debug: deyeidc.0 (179666) Connection closed

</body>
</html>`

Eelisglitchy, 2024-01-29:
Hm merkwürdig, woran kann das liegen?

raschy, 2024-01-29:
Das bedeutet für mich, dass der Adapter keine Daten bekommt.

Eelisglitchy, 2024-01-29:
Hab mal alles bis auf die beiden oberen Coils rausgenommen die sollten aufjedenfall gehen auch den 2. Registerbereich rausgenommen.

deyeidc.0
2024-01-29 12:10:55.144	debug	Connection closed

deyeidc.0
2024-01-29 12:10:55.142	debug	Connection closed/destroyed

deyeidc.0
2024-01-29 12:10:55.141	debug	Connection timeout

deyeidc.0
2024-01-29 12:10:45.047	debug	Connected to server

deyeidc.0
2024-01-29 12:10:44.995	debug	try to connect . . .

deyeidc.0
2024-01-29 12:10:44.995	debug	checkUserData is ready

deyeidc.0
2024-01-29 12:10:44.995	info	Retrieving data from the inverter will be done every 60 seconds

deyeidc.0
2024-01-29 12:10:44.994	debug	IP address [192.168.21.29] seems to be valid.

deyeidc.0
2024-01-29 12:10:44.984	info	starting. Version 0.0.14 in /opt/iobroker/node_modules/iobroker.deyeidc, node: v18.19.0, js-controller: 5.0.17

deyeidc.0
2024-01-29 12:10:41.046	info	Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason

deyeidc.0
2024-01-29 12:10:41.046	info	terminating

deyeidc.0
2024-01-29 12:10:41.046	debug	[onUnload] cleaned everything up...

deyeidc.0
2024-01-29 12:10:41.045	info	Got terminate signal TERMINATE_YOURSELF

Eelisglitchy, 2024-01-29:
Hm hab das auf meiner 2. Iobroker install im Garten mal probiert grade, da kommt das selbe raus.

raschy, 2024-01-29:
Möglicherweise kann eine direkte Installation via Experteneinstellung>Katze>Benutzerdefiniert>https://github.com/raschy/ioBroker.deyeidc  uns hier weiterhelfen.

Eelisglitchy, 2024-01-29:
Hab ich mal gemacht aber ich muss jetzt erstmal zur Arbeit. 

Er bleibt jetzt so stehen

deyeidc.0
2024-01-29 12:31:30.667	debug	Connected to server

deyeidc.0
2024-01-29 12:31:30.624	debug	try to connect . . .

deyeidc.0
2024-01-29 12:31:30.623	debug	checkUserData is ready

deyeidc.0
2024-01-29 12:31:30.623	info	Retrieving data from the inverter will be done every 60 seconds

deyeidc.0
2024-01-29 12:31:30.623	debug	IP address [192.168.21.29] seems to be valid.

deyeidc.0
2024-01-29 12:31:30.614	info	starting. Version 0.0.14 (non-npm: raschy/ioBroker.deyeidc#bace94a1df7286b00883044b02c6ea698b2f7c1c) in /opt/iobroker/node_modules/iobroker.deyeidc, node: v18.19.0, js-controller: 5.0.17

raschy, 2024-01-29:
So long.