raschy/ioBroker.deyeidc #81: Fehler bei der Installation CODE25

amnesia106, 2024-02-17:
Die installation von deyeidc endet mit dem Fehlercode 25. Anderer Adapter lassen sich ohne Probleme installieren.

jscontroller 5.0.19

`$ iobroker add deyeidc auto --host DietPi --debug
Installing iobroker.deyeidc@0.0.14... (System call)
npm ERR! code 1
npm ERR! path /opt/iobroker/node_modules/bufferutil
npm ERR! command failednpm ERR! command sh -c node-gyp-buildnpm ERR! gyp info it worked if it ends with oknpm ERR! gyp info using node-gyp@7.1.2npm ERR! gyp info using node@18.19.1 | linux | arm64npm ERR! gyp info find Python using Python version 3.11.2 found at "/usr/bin/python3"npm ERR! (node:3748) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.
npm ERR! (Use `node --trace-deprecation ...` to show where the warning was created)npm ERR! gyp info spawn /usr/bin/python3npm ERR! gyp info spawn args [npm ERR! gyp info spawn args '/opt/iobroker/node_modules/node-gyp/gyp/gyp_main.py',npm ERR! gyp info spawn args 'binding.gyp',npm ERR! gyp info spawn args '-f',npm ERR! gyp info spawn args 'make',npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/opt/iobroker/node_modules/bufferutil/build/config.gypi',npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/opt/iobroker/node_modules/node-gyp/addon.gypi',npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/home/iobroker/.cache/node-gyp/18.19.1/include/node/common.gypi',npm ERR! gyp info spawn args '-Dlibrary=shared_library',npm ERR! gyp info spawn args '-Dvisibility=default',npm ERR! gyp info spawn args '-Dnode_root_dir=/home/iobroker/.cache/node-gyp/18.19.1',npm ERR! gyp info spawn args '-Dnode_gyp_dir=/opt/iobroker/node_modules/node-gyp',npm ERR! gyp info spawn args '-Dnode_lib_file=/home/iobroker/.cache/node-gyp/18.19.1/<(target_arch)/node.lib',npm ERR! gyp info spawn args '-Dmodule_root_dir=/opt/iobroker/node_modules/bufferutil',npm ERR! gyp info spawn args '-Dnode_engine=v8',npm ERR! gyp info spawn args '--depth=.',npm ERR! gyp info spawn args '--no-parallel',npm ERR! gyp info spawn args '--generator-output',npm ERR! gyp info spawn args 'build',npm ERR! gyp info spawn args '-Goutput_dir=.'npm ERR! gyp info spawn args ]npm ERR! Traceback (most recent call last):npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/gyp_main.py", line 51, in <module>npm ERR! sys.exit(gyp.script_main())npm ERR! ^^^^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 670, in script_mainnpm ERR! return main(sys.argv[1:])npm ERR! ^^^^^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 662, in mainnpm ERR! return gyp_main(args)npm ERR! ^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 629, in gyp_mainnpm ERR! [generator, flat_list, targets, data] = Load(npm ERR! ^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 150, in Loadnpm ERR! result = gyp.input.Load(npm ERR! ^^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 3021, in Loadnpm ERR! LoadTargetBuildFile(npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 411, in LoadTargetBuildFilenpm ERR! build_file_data = LoadOneBuildFile(npm ERR! ^^^^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 239, in LoadOneBuildFilenpm ERR! build_file_contents = open(build_file_path, "rU").read()npm ERR! ^^^^^^^^^^^^^^^^^^^^^^^^^^^npm ERR! ValueError: invalid mode: 'rU' while trying to load binding.gypnpm ERR! gyp ERR! configure error npm ERR! gyp ERR! stack Error: `gyp` failed with exit code: 1npm ERR! gyp ERR! stack at ChildProcess.onCpExit (/opt/iobroker/node_modules/node-gyp/lib/configure.js:351:16)npm ERR! gyp ERR! stack at ChildProcess.emit (node:events:517:28)npm ERR! gyp ERR! stack at ChildProcess._handle.onexit (node:internal/child_process:292:12)npm ERR! gyp ERR! System Linux 6.1.21-v8+npm ERR! gyp ERR! command "/usr/bin/node" "/opt/iobroker/node_modules/.bin/node-gyp" "rebuild"npm ERR! gyp ERR! cwd /opt/iobroker/node_modules/bufferutilnpm ERR! gyp ERR! node -v v18.19.1npm ERR! gyp ERR! node-gyp -v v7.1.2npm ERR! gyp ERR! not ok
npm ERR! A complete log of this run can be found in: /home/iobroker/.npm/_logs/2024-02-17T06_56_31_890Z-debug-0.log
npm ERR! code 1npm ERR! path /opt/iobroker/node_modules/bufferutilnpm ERR! command failednpm ERR! command sh -c node-gyp-buildnpm ERR! gyp info it worked if it ends with oknpm ERR! gyp info using node-gyp@7.1.2npm ERR! gyp info using node@18.19.1 | linux | arm64npm ERR! gyp info find Python using Python version 3.11.2 found at "/usr/bin/python3"npm ERR! (node:3748) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.npm ERR! (Use `node --trace-deprecation ...` to show where the warning was created)npm ERR! gyp info spawn /usr/bin/python3npm ERR! gyp info spawn args [npm ERR! gyp info spawn args '/opt/iobroker/node_modules/node-gyp/gyp/gyp_main.py',npm ERR! gyp info spawn args 'binding.gyp',npm ERR! gyp info spawn args '-f',npm ERR! gyp info spawn args 'make',npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/opt/iobroker/node_modules/bufferutil/build/config.gypi',npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/opt/iobroker/node_modules/node-gyp/addon.gypi',npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/home/iobroker/.cache/node-gyp/18.19.1/include/node/common.gypi',npm ERR! gyp info spawn args '-Dlibrary=shared_library',npm ERR! gyp info spawn args '-Dvisibility=default',npm ERR! gyp info spawn args '-Dnode_root_dir=/home/iobroker/.cache/node-gyp/18.19.1',npm ERR! gyp info spawn args '-Dnode_gyp_dir=/opt/iobroker/node_modules/node-gyp',npm ERR! gyp info spawn args '-Dnode_lib_file=/home/iobroker/.cache/node-gyp/18.19.1/<(target_arch)/node.lib',npm ERR! gyp info spawn args '-Dmodule_root_dir=/opt/iobroker/node_modules/bufferutil',npm ERR! gyp info spawn args '-Dnode_engine=v8',npm ERR! gyp info spawn args '--depth=.',npm ERR! gyp info spawn args '--no-parallel',npm ERR! gyp info spawn args '--generator-output',npm ERR! gyp info spawn args 'build',npm ERR! gyp info spawn args '-Goutput_dir=.'npm ERR! gyp info spawn args ]npm ERR! Traceback (most recent call last):npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/gyp_main.py", line 51, in <module>npm ERR! sys.exit(gyp.script_main())npm ERR! ^^^^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 670, in script_mainnpm ERR! return main(sys.argv[1:])npm ERR! ^^^^^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 662, in mainnpm ERR! return gyp_main(args)npm ERR! ^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 629, in gyp_mainnpm ERR! [generator, flat_list, targets, data] = Load(npm ERR! ^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 150, in Loadnpm ERR! result = gyp.input.Load(npm ERR! ^^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 3021, in Loadnpm ERR! LoadTargetBuildFile(npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 411, in LoadTargetBuildFilenpm ERR! build_file_data = LoadOneBuildFile(npm ERR! ^^^^^^^^^^^^^^^^^npm ERR! File "/opt/iobroker/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 239, in LoadOneBuildFilenpm ERR! build_file_contents = open(build_file_path, "rU").read()npm ERR! ^^^^^^^^^^^^^^^^^^^^^^^^^^^npm ERR! ValueError: invalid mode: 'rU' while trying to load binding.gypnpm ERR! gyp ERR! configure error npm ERR! gyp ERR! stack Error: `gyp` failed with exit code: 1npm ERR! gyp ERR! stack at ChildProcess.onCpExit (/opt/iobroker/node_modules/node-gyp/lib/configure.js:351:16)npm ERR! gyp ERR! stack at ChildProcess.emit (node:events:517:28)npm ERR! gyp ERR! stack at ChildProcess._handle.onexit (node:internal/child_process:292:12)npm ERR! gyp ERR! System Linux 6.1.21-v8+npm ERR! gyp ERR! command "/usr/bin/node" "/opt/iobroker/node_modules/.bin/node-gyp" "rebuild"npm ERR! gyp ERR! cwd /opt/iobroker/node_modules/bufferutilnpm ERR! gyp ERR! node -v v18.19.1npm ERR! gyp ERR! node-gyp -v v7.1.2npm ERR! gyp ERR! not oknpm ERR! A complete log of this run can be found in: /home/iobroker/.npm/_logs/2024-02-17T06_56_31_890Z-de [... cut ...]
host.DietPi Cannot install iobroker.deyeidc@0.0.14: 1
ERROR: Process exited with code 25`

Node wurde komplett neu installiert.

`root@DietPi:~# uname -m && type -P nodejs node npm npx corepack && nodejs -v && node -v && npm -v && npx -v && corepack -v && iob -v && whoami && groups && echo $XDG_SESSION_TYPE && pwd && sudo apt update &> /dev/null && sudo apt update && apt policy nodejs
aarch64
/usr/bin/nodejs
/usr/bin/node
/usr/bin/npm
/usr/bin/npx
/usr/bin/corepack
v18.19.1
v18.19.1
10.2.4
10.2.4
0.22.0
5.0.12
root
root`

amnesia106, 2024-02-17:
€: node-gyp neu installiert und bufferutil nachgeschoben hat den Fehler behoben...

closed 2024-02-17 by amnesia106

raschy, 2024-02-17:
Ich war mir sehr sicher, dass es nicht an dem Adapter liegt.

Pocky2507, 2024-05-06:
> €: node-gyp neu installiert und bufferutil nachgeschoben hat den Fehler behoben...

@amnesia106 
Kannst Du mal erklären wie genau Du das gemacht hast.
Habe hier den gleichen Fehler.