raschy/ioBroker.deyeidc #87: Warning "invalid json config"

HalloNo, 2024-03-02:
Hallo Rasch,
ich habe den Deye Sun600 und versuche diesen über IOBroker auszulesen.
Die Installation habe ich nun schon mehrfach auf meine Raspi 3B+ durchgeführt, ohne Erfolg.

Ich hatte die GitHub Version als auch die freigebene Version probiert, ich kriege keine Daten vom Inverter.

Hier der letzte Log:

2024-03-02 15:53:50.662 - info: admin.0 (1329) ==> Connected system.user.admin from ::ffff:192.168.179.2
2024-03-02 15:55:57.703 - warn: admin.0 (1329) **deyeidc has an invalid jsonConfig:** [{"instancePath":"/items/_coils/items/coils/items/1","schemaPath":"#/items/allOf/17/then/additionalProperties","keyword":"additionalProperties","params":{"additionalProperty":"trim"},"message":"must NOT have additional properties"},{"instancePath":"/items/_coils/items/coils","schemaPath":"#/patternProperties/%5E.%2B/allOf/23/if","keyword":"if","params":{"failingKeyword":"then"},"message":"must match \"then\" schema"},{"instancePath":"/items/_coils","schemaPath":"#/properties/items/patternProperties/%5E.%2B/allOf/8/if","keyword":"if","params":{"failingKeyword":"then"},"message":"must match \"then\" schema"}]
2024-03-02 15:57:00.085 - info: host.raspiBDW "system.adapter.deyeidc.0" enabled
2024-03-02 15:57:00.176 - info: host.raspiBDW instance system.adapter.deyeidc.0 started with pid 1717
2024-03-02 15:57:03.558 - info: deyeidc.0 (1717) starting. Version 0.0.14 in /opt/iobroker/node_modules/iobroker.deyeidc, node: v19.9.0, js-controller: 5.0.19
2024-03-02 15:57:03.619 - info: deyeidc.0 (1717) Retrieving data from the inverter will be done every 60 seconds

Was mach ich falsch??

Grüße
Hans

raschy, 2024-03-04:
Gar nichts. Hier wird nur eine Warnung ausgegeben, dass eine Datei nicht (mehr) ganz korrekt ist. Sollte aber keine Auswirkungen auf die Funktion des Adapters haben. Wird aber nichts desto trotz beim nächsten Update korrigiert.
Wenn die IP und die Logger-Seriennummer korrekt sind, müssen Daten ankommen, zumindest die vom "Apo_t1" - gerade mit einem RPi4 getestet.

raschy, 2024-07-01:
No new findings. Therefore closed from my side.