raschy/ioBroker.deyeidc #88: How to write register: 40 Active Power Regulation?

fat-randy, 2024-03-18:
Hallo Raschy,

ich lese mit deinem Adapter erfolgreich zwei Deye Sun 600 Mikrowechselrichter aus. Funktioniert super.
Da meine WR über keine Modbus Schnittstelle verfügen würde ich auch gerne das Register: 40 Active Power Regulation damit beschreiben um eine Nulleinspeisung zu erreichen.
Leider speichert er den Wert nicht. Ich vermute ich beschreibe damit nur den Logger und der holt sich den Wert immer wieder neu vom Wechselrichter.
Gibt es da irgendeine Möglichkeit das mit deinem Adapter zu machen? Oder mit Modbus Adapter im iOBroker.
Die Lösung von: https://github.com/kbialek/deye-inverter-mqtt?tab=readme-ov-file#readme ist mir zu kompliziert.
Bitte um Antwort.
Gruß Manuel

raschy, 2024-03-18:
Hallo Manuel, Kannst du mir bitte mal schildern, was du bisher gemacht bzw. versucht hast. Gerne auch per Mail an raschy@gmx.de. Möglicherweise habe ich eine Lösung für dich.

closed 2024-03-22 by raschy

JoGu01, 2024-03-22:
Hi ihr beiden. Mag mir noch einer sagen wie ihr das gelöst habt?

raschy, 2024-03-22:
Mach mal den gleichen Weg wie fat-randy, per Mail.