raschy/ioBroker.deyeidc #94: Iobroker DEYE 12K LAN Adapter

Photographix, 2024-04-03:
**Describe the bug**  
Adapter liest im IObroker keine Werte aus

**To Reproduce**  
Steps to reproduce the behavior:
- Deye 12k -> **LAN** Adapter
- IObroker
- Adapter deyeidc v0.0.14
- json File from github (deyeidc.Sun12KP3.json)
- Settings wie IP, Pollimg und Port gesetzt

**Expected behavior**  
Der Adapter liest nach "Objekte" alle Werte vom Deye aus

**Screenshots & Logfiles**  
Fehler bei Protokolle Adapter deyeidc:
`Error: Frame CheckSum faulty! 8a|2a | Error: Frame CheckSum faulty! 8a|2a at idcCore.checkDataFrame (/opt/iobroker/node_modules/iobroker.deyeidc/lib/idc-core.js:70:10) at Deyeidc.onData (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:170:24) at Socket.<anonymous> (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:158:10) at Socket.emit (node:events:517:28) at addChunk (node:internal/streams/readable:368:12) at readableAddChunk (node:internal/streams/readable:341:9) at Readable.push (node:internal/streams/readable:278:10) at TCP.onStreamRead (node:internal/stream_base_commons:190:23)`

**Versions:**  
 - Adapter version: 0.0.14
- Plattform: docker (official image - v9.1.1)
- RAM: 15.3 GB
- Node.js: v18.19.1(18.20.0)
- NPM: 10.2.4(10.5.0)

raschy, 2024-04-03:
Ich vermute mal, dass es irgendein Firmware-Update gibt, mit dem die Checksumme nun anders berechnet wird. Wäre es ein genereller Fehler hätte sich schon mehr Nutzer melden müssen. Als erste Möglichkeit bitte mal den Adapter direkt von Github installieren.

Rene55-0, 2024-04-03:
Ist ein ganz anderes Fehlerbild. Ein anderer Nutzer hat dieses (mit beiden Fehlern) auch so erlebt. Er wartet gerade auf einen neuen WR, danach sehen wir ob es am Adapter (insbesondere bei den Großen) liegt oder sonstwo. Im Augenblick bin ich ratlos.

raschy, 2024-04-03:
Ist ein ganz anderes Fehlerbild. Ein anderer Nutzer hat dieses (mit beiden Fehlern) auch so erlebt. Er wartet gerade auf einen neuen WR, danach sehen wir ob es am Adapter (insbesondere bei den Großen) liegt oder sonstwo. Im Augenblick bin ich ratlos.

raschy, 2024-07-01:
No new findings. Therefore closed from my side.

closed 2024-07-01 by raschy