raschy/ioBroker.deyeidc #96: Fehlermeldung

Pocky2507, 2024-05-06:
Kann mir jemand erklären wie ich das weg bekomme?
`$ iobroker url https://github.com/raschy/ioBroker.deyeidc --host rockpro64 --debug

install raschy/ioBroker.deyeidc#c8e6ee63f93e3bc8f6bef811e9dc77a6061b1dcb

NPM version: 10.5.0

Installing raschy/ioBroker.deyeidc#c8e6ee63f93e3bc8f6bef811e9dc77a6061b1dcb... (System call)

npm WARN skipping integrity check for git dependency ssh://git@github.com/raschy/ioBroker.deyeidc.git

npm ERR! code 1

npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/opt/iobroker/node_modules/node-gyp/addon.gypi',npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/home/iobroker/.cache/node-gyp/18.20.2/include/node/common.gypi',npm ERR! gyp info spawn args '-Dlibrary=shared_library',npm ERR! gyp info spawn args '-Dvisibility=default',npm ERR! gyp info spawn args '-Dnode_root_dir=/home/iobroker/.cache/node-gyp/18.20.2',npm ERR! gyp info spawn args '-Dnode_gyp_dir=/opt/iobroker/node_modules/node-gyp',npm ERR! gyp info spawn args '-Dnode_lib_file=/home/iobroker/.cache/node-gyp/18.20.2/<(target_arch)/node.lib',npm ERR! gyp info spawn args '-Dmodule_root_dir=/opt/iobroker/node_modules/utf-8-validate',npm ERR! gyp info spawn args '-Dnode_engine=v8',npm ERR! gyp info spawn args '--depth=.',npm ERR! gyp info spawn args '--no-parallel',npm ERR! gyp info spawn args '--generator-output',npm ERR! gyp info spawn args 'build',npm ERR! gyp info spawn args '-Goutput_dir=.'npm ERR! gyp info spawn args ]npm ERR! gyp info spawn makenpm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]npm ERR! make: cc: No such file or directorynpm ERR! make: *** [validation.target.mk:111: Release/obj.target/validation/src/validation.o] Error 127npm ERR! gyp ERR! build error npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2npm ERR! gyp ERR! stack at ChildProcess.<anonymous> (/opt/iobroker/node_modules/node-gyp/lib/build.js:209:23)npm ERR! gyp ERR! System Linux 5.15.93-rockchip64npm ERR! gyp ERR! command "/usr/bin/node" "/opt/iobroker/node_modules/node-gyp/bin/node-gyp.js" "rebuild"npm ERR! gyp ERR! cwd /opt/iobroker/node_modules/utf-8-validatenpm ERR! gyp ERR! node -v v18.20.2npm ERR! gyp ERR! node-gyp -v v10.1.0npm ERR! gyp ERR! not oknpm ERR! A complete log of this run can be found in: /home/iobroker/.npm/_logs/2024-05-06T11_41_19_591Z-debug-0.log

npm WARN skipping integrity check for git dependency ssh://git@github.com/raschy/ioBroker.deyeidc.git npm ERR! code 1npm ERR! path /opt/iobroker/node_modules/utf-8-validatenpm ERR! command failednpm ERR! command sh -c node-gyp-buildnpm ERR! make: Entering directory '/opt/iobroker/node_modules/utf-8-validate/build'npm ERR! CC(target) Release/obj.target/validation/src/validation.onpm ERR! make: Leaving directory '/opt/iobroker/node_modules/utf-8-validate/build'npm ERR! gyp info it worked if it ends with oknpm ERR! gyp info using node-gyp@10.1.0npm ERR! gyp info using node@18.20.2 | linux | arm64npm ERR! gyp info find Python using Python version 3.10.12 found at "/usr/bin/python3"npm ERR! gyp info spawn /usr/bin/python3npm ERR! gyp info spawn args [npm ERR! gyp info spawn args '/opt/iobroker/node_modules/node-gyp/gyp/gyp_main.py',npm ERR! gyp info spawn args 'binding.gyp',npm ERR! gyp info spawn args '-f',npm ERR! gyp info spawn args 'make',npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/opt/iobroker/node_modules/utf-8-validate/build/config.gypi',npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/opt/iobroker/node_modules/node-gyp/addon.gypi',npm ERR! gyp info spawn args '-I',npm ERR! gyp info spawn args '/home/iobroker/.cache/node-gyp/18.20.2/include/node/common.gypi',npm ERR! gyp info spawn args '-Dlibrary=shared_library',npm ERR! gyp info spawn args '-Dvisibility=default',npm ERR! gyp info spawn args '-Dnode_root_dir=/home/iobroker/.cache/node-gyp/18.20.2',npm ERR! gyp info spawn args '-Dnode_gyp_dir=/opt/iobroker/node_modules/node-gyp',npm ERR! gyp info spawn args '-Dnode_lib_file=/home/iobroker/.cache/node-gyp/18.20.2/<(target_arch)/node.lib',npm ERR! gyp info spawn args '-Dmodule_root_dir=/opt/iobroker/node_modules/utf-8-validate',npm ERR! gyp info spawn args '-Dnode_engine=v8',npm ERR! gyp info spawn args '--depth=.',npm ERR! gyp info spawn args '--no-parallel',npm ERR! gyp info spawn args '--generator-output',npm ERR! gyp info spawn args 'build',npm ERR! gyp info spawn args '-Goutput_dir=.'npm ERR! gyp info spawn args ]npm ERR! gyp info spawn makenpm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]npm ERR! make: cc: No such file or directorynpm ERR! make: *** [validation.target.mk:111: Release/obj.target/validation/src/validation.o] Error 127npm ERR! gyp ERR! build error npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2npm ERR! gyp ERR! stack at ChildProcess.<anonymous> (/opt/iobroker/node_modules/node-gyp/lib/build.js:209:23)npm ERR! gyp ERR! System Linux 5.15.93-rockchip64npm ERR! gyp ERR! command "/usr/bin/node" "/opt/iobroker/node_modules/node-gyp/bin/node-gyp.js" "rebuild"npm ERR! gyp ERR! cwd /opt/iobroker/node_modules/utf-8-validatenpm ERR! gyp ERR! node -v v18.20.2npm ERR! gyp ERR! node-gyp -v v10.1.0npm ERR! gyp ERR! not oknpm ERR! A complete log of this run can be found in: /home/iobroker/.npm/_logs/2024-05-06T11_41_19_591Z-debug-0.log

host.rockpro64 Cannot install raschy/ioBroker.deyeidc#c8e6ee63f93e3bc8f6bef811e9dc77a6061b1dcb: 1

ERROR: Process exited with code 25`

raschy, 2024-05-31:
Ich habe bei dem Adapter keine besonderen Dinge eingestellt, um den zu bauen. Hat diese Vermutung etwas mit https://forum.iobroker.net/topic/74544/fehlermeldung-bei-adapter-installation zu tun? 
Kannst du mir den Umstand  zum C-Compiler näher erläutern? Muss nicht hier sein kann auch per mail - ich möchte einfach, dass der Adapter überall funktioniert!

Grothesk242, 2024-05-31:
Ich kann es auch nicht genau sagen. Dazu müsste man genaueres zu den verwendeten Systemen wissen. Also welche compiler da in welcher Version vorhanden sind. Hier läuft es jedenfalls mit gcc 14.0.1 durch :+1: 
```
$ iobroker url raschy/ioBroker.deyeidc --host chet --debug

install raschy/ioBroker.deyeidc
NPM version: 10.7.0
Installing raschy/ioBroker.deyeidc... (System call)
npm warn ERESOLVE overriding peer dependency
npm warn
npm warn Could not resolve dependency:
npm warn peer @iobroker/types@"^5.0.11" from @iobroker/adapter-core@3.1.4
npm warn node_modules/iobroker.admin/node_modules/@iobroker/adapter-core
npm warn @iobroker/adapter-core@"^3.1.4" from iobroker.admin@6.17.14
npm warn node_modules/iobroker.admin
npm warn
npm warn Conflicting peer dependency: @iobroker/types@5.0.19
npm warn node_modules/@iobroker/types
npm warn peer @iobroker/types@"^5.0.11" from @iobroker/adapter-core@3.1.4
npm warn node_modules/iobroker.admin/node_modules/@iobroker/adapter-core
npm warn @iobroker/adapter-core@"^3.1.4" from iobroker.admin@6.17.14
npm warn node_modules/iobroker.admin

189 packages are looking for funding run `npm fund` for details

upload [12] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/deyeidc.png deyeidc.png image/png
upload [11] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/i18n/de/translations.json i18n/de/translations.json application/json
upload [10] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/i18n/en/translations.json i18n/en/translations.json application/json
upload [9] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/i18n/es/translations.json i18n/es/translations.json application/json
upload [7] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/i18n/it/translations.json i18n/it/translations.json application/json
upload [5] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/i18n/pl/translations.json i18n/pl/translations.json application/json
upload [4] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/i18n/pt/translations.json i18n/pt/translations.json application/json
upload [3] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/i18n/ru/translations.json i18n/ru/translations.json application/json
upload [2] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/i18n/uk/translations.json i18n/uk/translations.json application/json
upload [1] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/i18n/zh-cn/translations.json i18n/zh-cn/translations.json application/json
upload [0] deyeidc.admin /opt/iobroker/node_modules/iobroker.deyeidc/admin/jsonConfig.json jsonConfig.json application/json

Process exited with code 0
```

Grothesk242, 2024-05-31:
Irgendwas klemmt da wohl bei utf-8-validate. Bei mir schaut das aber so aus: 
```
echad@chet:/opt/iobroker $ npm ls utf-8-validate
iobroker.inst@3.0.0 /opt/iobroker
├─┬ iobroker.admin@6.17.14
│ └─┬ ws@8.17.0
│   └── utf-8-validate@5.0.10
├─┬ iobroker.deyeidc@0.0.14 (git+ssh://git@github.com/raschy/ioBroker.deyeidc.git#43c686366918772d8a1316e0b6358f0fd24abdf1)
│ └─┬ websocket@1.0.35
│   └── utf-8-validate@5.0.10 deduped
├─┬ iobroker.govee-app@0.0.6 (git+ssh://git@github.com/TA2k/ioBroker.govee-app.git#6a0f84cfb34eb3a0ca647c7a519b9d5a48ccfb85)
│ └─┬ aws-iot-device-sdk@2.2.13
│   └─┬ mqtt@4.2.8
│     └─┬ ws@7.5.9
│       └── utf-8-validate@5.0.10 deduped
├─┬ iobroker.midea@0.1.0 (git+ssh://git@github.com/TA2k/ioBroker.midea.git#0d037c0c9a025b193546f881f747c8f998bfc724)
│ └─┬ pythonia@1.1.3
│   └─┬ ws@7.5.9
│     └── utf-8-validate@5.0.10 deduped
├─┬ iobroker.samsung_tizen@1.1.0
│ └─┬ ws@7.5.9
│   └── utf-8-validate@5.0.10 deduped
├─┬ iobroker.socketio@6.7.0
│ └─┬ socket.io@2.5.0
│   ├─┬ engine.io@3.6.1
│   │ └─┬ ws@7.4.6
│   │   └── utf-8-validate@5.0.10 deduped
│   └─┬ socket.io-client@2.5.0
│     └─┬ engine.io-client@3.5.3
│       └─┬ ws@7.4.6
│         └── utf-8-validate@5.0.10 deduped
└─┬ iobroker.tuya@3.15.0
  └─┬ mqtt@4.3.8
    └─┬ ws@7.5.9
      └── utf-8-validate@5.0.10 deduped

echad@chet:/opt/iobroker $ 

```

raschy, 2024-05-31:
Ich bin jetzt ziemlich verunsichert: Ist gcc nicht ein Compiler für c++Programme? Wo wird der im ioBroker-Umfeld genutzt? Bin ich bzw. der Adapter dafür zuständig, den Compiler, wenn er denn gebraucht wird, mit zu installieren.?
Und beim utf-8 validator bin ich auch überfordert. Da höre ich gerne auf weitere Infos.

Grothesk242, 2024-06-01:
gcc ist eine Sammlung von mehreren Compilern. Weil der 'überall' verwendet wird zieht der ioBroker-Installer das entsprechende Paket 'build-essential' ohnehin rein. Vielleicht beim TE in einer älteren Version. Sein
System Linux 5.15.93-rockchip64
dürfte ja etwas angejahrt sein.

raschy, 2024-06-01:
Konkret habe ich an dem Adapter schon länger nichts mehr gemacht, außer eben die üblichen monatlichen Dependabot-Updates. Wenn ich das jetzt richtig verstehe, bin ich hier eh außen vor, und der User müsste sein System auf den neuesten Stand bringen. Right?

Grothesk242, 2024-06-01:
Ich vermute. Man müsste mal die genauere Umgebung kennen, die da vorhanden ist. Auf aktuellen Kisten baut der Adapter ja inkl. aller Deps, wie man oben sieht.

raschy, 2024-06-01:
Somit entweder gelöst oder für den TE nicht so wichtig. Danke Dir. 
Dann mach ich hier mal zu.

closed 2024-06-01 by raschy