raschy/ioBroker.deyeidc #103: Serial with characters

kblumenau, 2024-06-07:
I have two deye inverters up and running with v0.0.14 in iobroker. 

I now got a new one (M80G4) that has a serial number including a character at the end - this one is not accepted by the deyeidc adapter
other than that, the inverter looks the same compared to the other two

can this be adapted / will that one work?

thanks for your work and support!

raschy, 2024-06-07:
I have always assumed that serial numbers are actually numbers. Which symbol is on the back of your M80G4?

kblumenau, 2024-06-07:
Thanks for your quick answer!

What symbol do you mean? we can share more details through private msg if you want...
Serial printed on the back is a 9 digit number plus an F at the end
inverter is online and connected to the cloud, I added  it to the SolarPV config, serial was accepted but I do not see data yet (not connected). I also tried some of the other numbers I could find without success. 
As said, the other two old inverters work (local and cloud), so I thought I knew what I should do... but...

raschy, 2024-06-07:
That was the question about the ‘F’. However, we can also exchange details by e-mail (also in German).

kblumenau, 2024-06-07:
blume@gmx.li

kblumenau, 2024-06-07:
SORRY! 
it was just my fault, using the wrong number. Using the correct number ("device serial number" on web interface) the adapter works perfectly, just as is did before. Thanks again for the work and support ... and sorry

closed 2024-06-07 by raschy