raschy/ioBroker.deyeidc #109: DynDNS IP not possibe

radebold, 2024-07-09:
Hi,

it is not possible to use dyndns?
IP address [#####.ddns.net] is not valid !

it is possible to add this support?

Thx.
Thomas

raschy, 2024-07-11:
Hi Thomas, what is the use case for this extension? I have installed this, please install from Github, test and give feedback.

radebold, 2024-07-11:
Hi rasche

I extended the IP Check, so it is also possible to put in a dyndns address
like xyz.dyndns.org

Thx

raschy, 2024-07-11:
Yes, I already understood the use. What is it used for/when? And, do you need any further forwarding? 
It's probably not enough if the public IP is recognized.

radebold, 2024-07-11:
My use case I montitoring the inverter of my mum remotely with my Iobroker

raschy, 2024-08-01:
Should be fixed with version 0.0.15 (and later).

closed 2024-08-01 by raschy