raschy/ioBroker.deyeidc #111: deyeidc has an invalid jsonConfig

mcm1957, 2024-07-21:
**Describe the bug**  
deyeidc has an invalid jsonConfig: [{"instancePath":"/items/_coils/items/coils/items/1","schemaPath":"#/items/allOf/17/then/additionalProperties","keyword":"additionalProperties","params":{"additionalProperty":"trim"},"message":"must NOT have additional properties"},{"instancePath":"/items/_coils/items/coils","schemaPath":"#/patternProperties/%5E.%2B/allOf/25/if","keyword":"if","params":{"failingKeyword":"then"},"message":"must match \"then\" schema"},{"instancePath":"/items/_coils","schemaPath":"#/properties/items/patternProperties/%5E.%2B/allOf/8/if","keyword":"if","params":{"failingKeyword":"then"},"message":"must match \"then\" schema"},{"instancePath":"","schemaPath":"#/if","keyword":"if","params":{"failingKeyword":"then"},"message":"must match \"then\" schema"}]

**To Reproduce**  
Steps to reproduce the behavior:
1. open config UI
2. check log

**Expected behavior**  
No error message

**Versions:**  
 - Adapter version: 0.0.14
 - JS-Controller version: n/a
 - Node version: n/a
 - Operating system: n/a

**Additional context**  
Error is caused by 
```
	{
                "type": "number",
		"trim": false,
		"attr": "rules",
		"width": "15%",
		"title": "rule",
		"default": ""
	}
```

Error seems to be fixed at github already but fix did not yet make its way into a release.

See https://forum.iobroker.net/topic/63899/adapter-cloudfreie-auslesung-von-deye-invertern/931

raschy, 2024-07-21:
Thanks for the tip, I'm just about to make a few more improvements to the adapter. A new version will then be created.

unstressable, 2024-07-24:
same problem here.

raschy, 2024-07-24:
This is not a problem - just a warning. As already written, will be fixed soon.

raschy, 2024-08-01:
Should be fixed with version 0.0.15 (and later).

closed 2024-08-01 by raschy