raschy/ioBroker.deyeidc #114: Seit V0.15 TypeError: Cannot read properties of undefined (reading 'register')

Baenker, 2024-07-30:
Seit der Installation der neuen Version 0.15 habe ich auf einmal diese Fehlermeldung im Log:

`TypeError: Cannot read properties of undefined (reading 'register') | TypeError: Cannot read properties of undefined (reading 'register') at Deyeidc.onData (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:166:11) at Socket.<anonymous> (/opt/iobroker/node_modules/iobroker.deyeidc/main.js:152:10) at Socket.emit (node:events:517:28) at addChunk (node:internal/streams/readable:368:12) at readableAddChunk (node:internal/streams/readable:341:9) at Readable.push (node:internal/streams/readable:278:10) at TCP.onStreamRead (node:internal/stream_base_commons:190:23) `

Ist das ein generelles Problem oder ein Einzelproblem was ich selber beheben kann?

Mit der alten Version war mein Log komplett unauffällig ohne Fehlermeldungen

raschy, 2024-07-30:
Entweder direkt von Github installieren oder morgen auf die 0.0.16 warten.

Baenker, 2024-07-30:
Vielen Dank für die schnelle Hilfe. Ich habe von Github installiert. Ich schließe das Ticket. Sollte es morgen noch Fehler melden würde ich mich wieder melden.

closed 2024-07-30 by Baenker