raschy/ioBroker.deyeidc #120: Verbindungsfehler

sonny1985, 2024-08-03:
Hallo.

Ich bleib derweil bei 0.0.14 . Mit der 0.0.15er bekam ich tonnenweise irgendwelche Verbindungsfehler , rot markiert, ins Log. Mit der 0.0.16er waren sie immer noch da, aber weniger (EHOST_unreach).

Ich installier schnell die 0.16er, und reiche einen Screenshot nach :)

LG

sonny1985, 2024-08-03:
[image]

OK Ich bins nicht. :) 
In der instanzenansicht wird der betroffene WR dann "orange"- Starte ich die Instanz neu, ist er gleich wieder grün. Ein Update macht er alle 310 sekunden...
Ich hab 2 Instanzen am laufen.

raschy, 2024-08-03:
Produzieren die WR ins Netz? Wenn nicht, kann keine Verbindung zum WR aufgebaut werden und die Instanz bleibt gelb.

sonny1985, 2024-08-03:
Hi.
Klar sind beide Aktiv AC Seitig.
Mach ich , ich schalt ihn auf Debug, und schick in ner halben Stunde nen Screenshot nach :)

raschy, 2024-08-03:
Ich weiß jetzt nicht, welchen WR du hast. Es gibt ein Verbindungstimeout von 10 Sekunden, welches die Verbindung nach dem Abholen - da nicht mehr gebraucht - schließt. Sollte das ein Problem werden, lass ich die Verbindung in der nächsten Version halt auf. Bitte Beobachten und berichten.

sonny1985, 2024-08-03:
Ich hab einen Sun-300 und einen Sun-1600 in Verwendung.

Ich beobachte das einfach ein wenig, ich halt dich am laufenden, es kann allerdings durchaus sein, das es an meinem Netzwerk lag. (wer weiß). 
War das mit der Verbindung in der Version 0.0.14 gleich ? (Die hat nie irgendwelche Roten Fehlereinträge verursacht)
Die 0.0.15 war mit roten Fehlern am schlimmsten.

raschy, 2024-08-03:
Ja, die 0.015 war etwas buggy. Da hatte ich zuviel wegoptimiert. 
Haben deine beiden SUN auch soetwas wie eine externe Relaisbox?

sonny1985, 2024-08-03:
Nein, die relais hab ich nicht installiert 😊.  Mit denen geht generell garnixmehr.

raschy, 2024-08-03:
Ich habe damals den Adapter so angepasst, dass die Kommunikation mit der Relaisbox rausgefiltert wird. Aber lass so wie es ist, generell eine Fehlerquelle weniger. Jetzt müssen wir nur noch schauen was orange/rot macht. Einfach wieder melden, Danke.

sonny1985, 2024-08-03:
[image]
[image]

Ok, an mir liegts nicht :) 
LG

sonny1985, 2024-08-03:
[image]

Das hier ist mir noch aufgefallen wenn ich zwischen verschiedenen Adapterversionen (0.0.14 und 0.0.16) wechsle.

raschy, 2024-08-03:
Versuch mal (morgen) direkt von Github zu installieren,

sonny1985, 2024-08-04:
Guten morgen. Habs mal direkt von hier installiert. 
Hab gesehen, du hast das Timeout auskommentiert, und die Verbindung nun immer stehen lassen oder ?

Ich meld mich am Abend, wenn beide WR vom Netz getrennt und offline sind, nochmals ob es Fehlereinträge im Log gibt :) Bis jetzt sieht das prima aus !.
LG und schönen Sonntag.

sonny1985, 2024-08-05:
Soo.

Das Log meldet mir am Tag bei beiden Instanzen , 4 Unterbrechungen. 
Wobei die Daten allerdings Lückenlos sind, seh ich in der Datenbank :)

Wenn das nun das gleiche ist vom Verbinden her wie die 0.0.14, versteh ich nicht, warum die 0.0.14 niemals einen einzigen Verbindungsfehler ins Log geschrieben hat .

Für mich ists nicht nachvollziehbar .
LG derweil

raschy, 2024-08-05:
Bei mir läuft die 0.0.16 auch gegen zwei WR. Auf beiden Systemen habe ich keine einzige (unerwartete) Unterbrechung. Jetzt wird es für mich schwierig.

sonny1985, 2024-08-05:
Eigenartig ist das wirklich, ich versteh es auch nicht.
Ich lass die 0.0.16 (Die Github version von hier) einfach mal drauf, und beobacht das einfach weiter.

raschy, 2024-08-05:
Ja, mach das so. Beobachte weiter und berichte, sobald du erkennen kannst, wann das genau passiert. Vllt. kann ich dann noch etwas verbessern.

sonny1985, 2024-08-08:
So , hatte die letzten 2 Tage keine roten Fehlermeldungen mehr im Log drin :)
Ich denk das kann man hier schließen. Bitte Frag mich nicht ob nun das aufrechterhalten der Verbindung dafür verantwortlich ist, oder da etwas anderes am werkeln war was für Probleme gesorgt hatte.

Hab auch bereits die 0.1.0er drauf :)

LG und dank.

closed 2024-08-27 by raschy