raschy/ioBroker.deyeidc #121: Deye Sun-12k sg04lp3 EU Register schreiben

DpunktS, 2024-08-06:
Ich besitze 2 Deye Sun-12k sg04lp3 EU Wechselrichter und lese mit deinem Adapter diese Wechselrichter in 10s Takt aus.

Ist es möglich mit deinem Adapter einen Schreibzugriff zu ermöglichen?

raschy, 2024-08-06:
Die Anfrage kam schon öfter. Die Möglichkeiten verschiedene Register beschreiben zu wollen ist recht umfangreich. Da müsste ich quasi das gleiche, was ich zum auslesen habe nochmals entwickeln. Da ich keine Möglichkeit habe zu testen, kann ich diese 'Schreibfunktion' nicht einbauen. Ich verweise hier gerne auf den 'Modbus-Adapter'- ob es damit funktioniert hat aber noch keiner zurückgemeldet.

crazy-ahmet, 2024-08-09:
> Ich besitze 2 Deye Sun-12k sg04lp3 EU Wechselrichter und lese mit deinem Adapter diese Wechselrichter in 10s Takt aus.
> 
> Ist es möglich mit deinem Adapter einen Schreibzugriff zu ermöglichen?

Bringt eine 10s Abfrage überhaupt was? Ich dachte der Logger aktualisiert die Daten sowieso nur alle Minute und in die Cloud schickt er Sie alle 5min. Wenn er intern nur alle 1min aktualisiert bringen doch 10s Abfragen nix oder ändern sich da tatsächlich Werte?

DpunktS, 2024-08-09:
Bei dem Deye sun-12k-sg04lp3-eu habe ich alle 10s tatsächlich andere Werte erhalten.

raschy, 2024-08-09:
Nach meinem Wissen haben die "großen" Wechselrichter einen richtigen Zugang zum Modbus. Daher sollte es doch ein leichtes sein, mit dem Modbus-Adapter die gewünschten Register zu beschreiben.

DpunktS, 2024-08-10:
Ja das ist richtig mit der Modbus Schnittstelle. Aktuell lese ich die Wechselrichter auch so aus.
In Zukunft wird aber diese Schnittstelle aufgrund des §14a EnWG höchstwahrscheinlich mit einer Steuerbox (IVG NetSync) belegt sein um die 4,2 kW Drosselung und die 25 kWp Begrenzung realisieren zu können.

crazy-ahmet, 2024-08-11:
> Ja das ist richtig mit der Modbus Schnittstelle. Aktuell lese ich die Wechselrichter auch so aus.
Du ließt mit einem Modbus Adapter aus oder dem Deyeidc hier? Habe auch ewig mit Modbus TCP Konvertern und anderen Github Projekten gebastelt bis ich wieder hier gelandet bin. Hier brauche ich ja keine zusätzliche Hardware in den Inverter basteln und es funktioniert, anscheinend auch im Minutentakt und weniger, ich dachte vorher immer der ließt das nur aus der Cloud alle 5min was mir wieder zu wenig war.

> In Zukunft wird aber diese Schnittstelle aufgrund des §14a EnWG höchstwahrscheinlich mit einer Steuerbox (IVG NetSync) belegt sein um die 4,2 kW Drosselung und die 25 kWp Begrenzung realisieren zu können.
Das ist hier nicht unbedingt nötig. Die neueren Geräte dieses Modells haben eine DRM Schnittstelle. Ist ein zusätzlicher RJ45 Port. Durch brücken verschiedener Kontakte kann die Leistung von 100%, 60%, 30% und 0% geregelt werden. Leider hat das nur der hier, ein simpler 25kW Deye String Inverter hat den Port nicht und gerade hier wäre er nötiger gewesen.

raschy, 2024-08-11:
Über wieviele Steuerungspunke (=Register) reden wir denn hier? Wenn das nur ein einziges Register ist, könnt ich mir das ja mal durch den Kopf gehen lassen.

DpunktS, 2024-08-11:
Aktuell verwende ich diese ESPHome Konfiguration für meine 2 Wechselrichter.
https://github.com/TheSmartGerman/esphome-for-deye/blob/main/deye12k%20esphome%20config

Es wäre noch schön, wenn man in deinen Adapter mehrere Geräte einfügen kann und diese Werte beide Geräte miteinander verrechnen kann. Grund dafür ist, wenn man z.B. 2 Wechselrichter in einem Master/Slave Verbund betreibt werden alle Werte unter den Geräten aufgeteilt (Leider auch die von den CT-Klemmen die nur an den Master angeschlossen sind). Im Master/Slave Verbund muss aber nur der Master beschrieben werden und der Slave übernimmt die Werte automatisch.

raschy, 2024-08-11:
Ich glaube nicht, dass ich in dem Adapter zwei Connections verwalten kann. Von daher bleibt das ein frommer Wunsch.

closed 2024-08-27 by raschy