raschy/ioBroker.deyeidc #122: Rechnungsfehler im Compute

zrockyy, 2024-08-11:
Moin.
Bei mir sind die Berechnungen im Compute falsch.
Als Beispiel:
PV1_Leistung+PV2_Leistung (PV_Gesamtleistung)
PV1_Leistung: 8000W
PV2_Leistung: 4000W
Nun sollte man glauben er spuckt  12000W aus, Falsch! es sind 80000000W
Das heißt er multipliziert PV1_Leistung mit 10000 und setzt das ergebnis als PV_Gesamtleistung

Selben Fehler habe ich bei Hausverbrauch.
Kann mir das einer begründen wieso er nicht pv1 und pv2 richtig addieren kann? Die Werte sind so komplett nutzlos, würde er wenigstens 12000000W ausgeben, könnte ich das ja /10000 teilen

Ebenso sind die ganzen GRID Werte falsch. ~63700W - Aber die logge ich nicht.

raschy, 2024-08-11:
Mag sein: _Die Werte sind so komplett nutzlos,_ Zeig mal deine Formel und stell bitte mal den Adapter auf debug. Dann sehen wir weiter.

zrockyy, 2024-08-12:
deyeidc.0 | 2024-08-12 10:10:00.047 | debug | [computeData]  {"values":"PV1_PV2_Leistung","key":"PV_Gesamt-Cluster","name":"PV_Gesamt-Cluster","unit":"W","factor":0}
-- | -- | -- | --
deyeidc.0 | 2024-08-12 10:10:00.047 | debug | [computeData]    {"values":"Day_PV_Power_Wh-Day_GridSell_Power_Wh","key":"Day_Eigenverbrauch","name":"Day  Eigenverbrauch","unit":"kWh","factor":0}
deyeidc.0 | 2024-08-12 10:10:00.047 | debug | [computeData]  {"values":"P_BACKUP_LOAD_TOTAL","key":"Hausverbrauch","name":"Hausverbrauch","unit":"W","factor":0}
deyeidc.0 | 2024-08-12 10:10:00.047 | debug | [computeData]    {"values":"PV1_PV2_Leistung+P_GEN_TOTAL","key":"PV_Gesamtleistung","name":"PV  - Gesamtleistung","unit":"W","factor":0}
deyeidc.0 | 2024-08-12 10:10:00.047 | debug | [computeData]  {"values":"PV1_Leistung+PV2_Leistung","key":"PV1_PV2_Leistung","name":"PV1+PV2 - Leistung","unit":"W","factor":0}
deyeidc.0 | 2024-08-12 10:10:00.047 | debug | Data reception for 6 registersets completed

Es sind die Standardformeln, ich habe selber keine Erstellt.

PV1_Leistung+PV2_Leistung = PV1_PV2_Leistung (Fehlerhaft)
PV1_PV2_Leistung+P_GEN_TOTAL = PV_Gesamtleistung (fehlerhaft)

Deshalb verstehe ich das nicht. PV1_Leistung ist korrekt, PV2_leistung ist korrekt.
Wenn ich PV1_Leistung+PV2_Leistung ist es falsch. da muss ja die ganze mathematik nicht stimmen.

Ich habe Version 0.16 (update am Samstag) hier steht aber es gibt 0.1.0 - Ich weiß aber nicht wie ich das installieren kann. Bin jetzt nicht so der erfahrene Linux User.

raschy, 2024-08-12:
Na, so ganz korrekt ist die Formel **[computeData] {"values":"PV1_PV2_Leistung"**, ja nicht. Wo ist denn hier die Berechnungsvorschrift heist das **+** ?? Hat sich die Debug-Ausgabe verscluckt?

zrockyy, 2024-08-13:
Also da stand ein +0 und bei dem anderen glaube ich ein *2 hinter. Das hatte ich rausgenommen um zu schauen ob der Wert dadurch korrekt wird. - War nicht der Fall. Keine Änderung.
Daher hatte ich dann hier aufgemacht. Aber es wurde ja durch das Update gehoben, ich kann mir selber nicht erklären was da schief gelaufen sein kann.

raschy, 2024-08-13:
Gut zu wissen. Das Compute-Modul ist erst ab der Version 0.0.16 in der Lage, mit numerischen Werten, die nicht aus den Datenpunkten herrühren, umzugehen. Jetzt wäre auch ein "DV1*-.1" möglich, wenn man es denn braucht. Datenpunkte, die nicht vorhanden sind, geben eine Warnung aus.

MarAlMe, 2024-08-14:
Das Problem besteht definitiv seit dem letzten Update. 
@raschy die Werte stammen doch aus der Config-Vorlage hier im Repo. Ich verwende diese ebenfalls, nur etwas auf meinen Sun5K angepasst.

raschy, 2024-08-14:
Ich habe es gerade noch einmal getestet: 
[image] und hat richtig gerechnet.

MarAlMe, 2024-08-14:
Da gerade keine Sonne scheint, habe ich die Werte Battery_Power und Total_Load genommen. Das Ergebnis ist allerdings das Selbe... Der errechnete Wert ist viel zu hoch:

[image]

[image]

raschy, 2024-08-14:
Welche Version ist das?  Richtig addieren kann der Adapter (leider) erst ab 0.1.0.

MarAlMe, 2024-08-15:
Da ich gerade auf einen neuen Homeserver umziehe und dafür eine komplett neue ioBroker Instanz aufgesetzt habe, war ich der Meinung, dass die aktuellste Version 0.1.0 ist. Ich bekomme allerdings nur 0.0.16 installiert. Version 0.1.0 ist also noch nicht über das offizielle Repo verfügbar, richtig? 

Da nun beide Instanzen parallel laufen, habe ich diese direkt nebeneinander gehalten:

[image]

Hier ist sehr schön zu sehen, dass das Problem erst mit der Version 0.0.16 auftritt. Davor hatte die Berechnung korrekt funktioniert.

zrockyy, 2024-08-15:
Genau. Ich hatte auch 0.0.14 oder so.. habe dann Kumpel iobroker gezeigt und Update gemacht.
Dann war die Berechnung Schrott.
Hab dann Update über GitHub gemacht und es funktioniert wieder.

raschy, 2024-08-15:
In der Version 0.016 wurde das Compute-Modul komplett überarbeitet. Da ist dann wohl die "String-Addition" reingerutscht.
Die 0.1.0 ist noch etwas zu jung für Stable. Wird aber bald verfügbar sein.

sventy4, 2024-08-18:
Ich musste meinen iobroker neu aufsetzen und habe nun einen aktuellen iobroker mit deyeidc in Version 0.1.0. Bei mir tauchen ebenfalls die Fehlermeldungen "Compute Key 'DV1' not found!" auf; leider werden aber auch keine Datenpunkte angelegt.

[image]

Ich habe einen SUN300G3-EU-230 und habe es auch mit dem Import der deyeidc.M80G3.json probiert. Selbes Ergebnis.

Das erhalte ich als debug-Meldungen:
```
2024-08-18 12:32:58.554 - debug: deyeidc.0 (2046) Redis Objects: Use Redis connection: 127.0.0.1:9001
2024-08-18 12:32:58.563 - debug: deyeidc.0 (2046) Objects client ready ... initialize now
2024-08-18 12:32:58.564 - debug: deyeidc.0 (2046) Objects create System PubSub Client
2024-08-18 12:32:58.564 - debug: deyeidc.0 (2046) Objects create User PubSub Client
2024-08-18 12:32:58.569 - debug: deyeidc.0 (2046) Objects client initialize lua scripts
2024-08-18 12:32:58.571 - debug: deyeidc.0 (2046) Objects connected to redis: 127.0.0.1:9001
2024-08-18 12:32:58.578 - debug: deyeidc.0 (2046) Redis States: Use Redis connection: 127.0.0.1:9000
2024-08-18 12:32:58.580 - debug: deyeidc.0 (2046) States create System PubSub Client
2024-08-18 12:32:58.580 - debug: deyeidc.0 (2046) States create User PubSub Client
2024-08-18 12:32:58.582 - debug: deyeidc.0 (2046) States connected to redis: 127.0.0.1:9000
2024-08-18 12:32:58.618 - info: deyeidc.0 (2046) starting. Version 0.1.0 in /opt/iobroker/node_modules/iobroker.deyeidc, node: v20.16.0, js-controller: 6.0.9
2024-08-18 12:32:58.624 - debug: deyeidc.0 (2046) IP address [192.168.178.12] seems to be valid.
2024-08-18 12:32:58.625 - info: deyeidc.0 (2046) Retrieving data from the inverter will be done every 360 seconds
2024-08-18 12:32:58.625 - debug: deyeidc.0 (2046) checkUserData is ready
2024-08-18 12:32:58.625 - debug: deyeidc.0 (2046) try to connect . . .
2024-08-18 12:32:58.631 - debug: deyeidc.0 (2046) Connected to server
2024-08-18 12:32:58.642 - debug: deyeidc.0 (2046) Response: (payload) {"register":1,"modbus":{"type":"Buffer","data":[6]}}
2024-08-18 12:32:58.657 - debug: deyeidc.0 (2046) Response: (payload) {"register":2,"modbus":{"type":"Buffer","data":[6]}}
2024-08-18 12:32:58.669 - debug: deyeidc.0 (2046) Response: (payload) {"register":3,"modbus":{"type":"Buffer","data":[6]}}
2024-08-18 12:32:58.699 - debug: deyeidc.0 (2046) Response: (payload) {"register":4,"modbus":{"type":"Buffer","data":[6]}}
2024-08-18 12:32:58.699 - debug: deyeidc.0 (2046) Data reception for 4 registersets completed
2024-08-18 12:32:58.700 - debug: deyeidc.0 (2046) [computeData] {"values":"DV1*DC1","key":"DP1","name":"DC Power PV1","unit":"W","factor":0}
2024-08-18 12:32:58.700 - warn: deyeidc.0 (2046) Compute Key 'DV1' not found!
2024-08-18 12:32:58.700 - debug: deyeidc.0 (2046) [computeData] {"values":"DV2*DC2","key":"DP2","name":"DC Power PV2","unit":"W","factor":0}
2024-08-18 12:32:58.700 - warn: deyeidc.0 (2046) Compute Key 'DV2' not found!
2024-08-18 12:32:58.700 - debug: deyeidc.0 (2046) [computeData] {"values":"DV3*DC3","key":"DP3","name":"DC Power PV3","unit":"W","factor":0}
2024-08-18 12:32:58.700 - warn: deyeidc.0 (2046) Compute Key 'DV3' not found!
2024-08-18 12:32:58.700 - debug: deyeidc.0 (2046) [computeData] {"values":"DV4*DC4","key":"DP4","name":"DC Power PV4","unit":"W","factor":0}
2024-08-18 12:32:58.700 - warn: deyeidc.0 (2046) Compute Key 'DV4' not found!
2024-08-18 12:32:58.700 - debug: deyeidc.0 (2046) [computeData] ResultJson: []
2024-08-18 12:32:58.701 - debug: deyeidc.0 (2046) [subscribeWatchpoint] set to 2301126511.Power_Set
2024-08-18 12:32:58.861 - debug: deyeidc.0 (2046) Response: (checkOnlineDate) {"register":0,"modbus":{"type":"Buffer","data":[6]}}
```

Hat jemand eine Idee, was ich falsch mache und warum keine Daten ankommen? Und woher weiß ich, welche Register der Inverter hat, hzw. welche Konfig zu benutzen ist? Ich habe zu meinem leider keinerlei Doku gefunden.

raschy, 2024-08-18:
Nunja, das ist ja kein Rechenfehler! Wenn es den 'DV1' nicht gibt, kann er auch nicht rechnen. So wie das aussieht, sind entweder keine Coils angelegt. _{"register":1,"modbus":{"type":"Buffer","data":[6]}}_ oder (wahrscheinlicher) ist die Seriennummer nicht korrekt. Die Datei für den M80G3 ist nach meiner Meinung korrekt. Bitte diese beiden Punkte prüfen. Dann wird der Adpater beim nächsten Datenabruf den Datenpunkt DV1 usw. anlegen und es wird laufen.

sventy4, 2024-08-18:
Danke, @raschy. Ich verstehe, dass es kein Rechenfehler ist, sondern ein Symptom. Ich prüfe morgen gerne nochmals die Seriennummer. Coils sind da.

closed 2024-10-12 by raschy