raschy/ioBroker.deyeidc #123: Deye 12K mit LAN Adapter

Photographix, 2024-08-19:
**Describe the bug**  
Verbindung scheint ok zu sein - bekomme keine Ergebnisse und auch keine weiteren Log Ausgaben. Es passiert einfach nichts. Instanz ist grün., im Objektbaum aber nichts.

**Expected behavior**  
Bei bestehendem Connect und ggf Fehlern - detaillierte Log Ausgabe. Logmodus "silly"

**Screenshots & Logfiles**  
[00340435095452027827](http://nolp.dhl.de/nextt-online-public/set_identcodes.do?lang=de&idc=00340435095452027827&rfn=&extendedSearch=true)

**Versions:**  
 - Adapter version: 0.1.0
 - JS-Controller version: 6.0.9
 - Node version: 20.16.0
 - Operating system: Docker Container

**Additional context**  
 - Ich nutze den **LAN** Logger und nicht den WLAN Logger
 - Logger Firmware **MW4C_14_CSIP_1.14**
 - Wechselrichter Deye **SUN 12K-SG04LP3-EU**
 - JSON File von hier (**deyeidc.Sun12KP3.json**)

raschy, 2024-08-19:
Leider kann ich mit dem o.g. Link nichts anfangen. Da kommt nur die DHL-Seite

Photographix, 2024-08-19:
Sorry ich hatte die falsche Zwischenablage. Oben und auch hier nochmal das aktuelle Log.

[image]

raschy, 2024-08-19:
Dieses Symptom habe ich heute (bei anderen Usern) auch schon gesehen. Bei mir mit zwei WR getestet ohne Fehler. 
Nur zum Vertsändnis für mich: was ist der Unterschied "Ich nutze den LAN Logger und nicht den WLAN Logger". Über welchen Weg die Daten ins Heimnetz kommen sollte doch eigentlich egal sein. Hauptsache stabil.

Photographix, 2024-08-19:
Ich war mir eben nicht sicher wie der Datenweg über den Logger geht. 90% der User haben sicher den dazugehörigen WLAN Logger. Ich war mir nicht sicher ob das ein hilfreicher Hinweis ist. Ich hatte auch bereits mit dem alten Adapter 0.14 keine Ergebnisse. Der LAN Logger ist zumindest bei mir stabiler als mit dem WLAN Logger.

raschy, 2024-08-27:
@Photographix Problem gelöst ?

Photographix, 2024-08-27:
Nein Problem besteht weiterhin. Habe aber auch nichts weiter versucht, da es ja nach dem connect keine weiteren Ausgaben gibt und so die Fehlersuche schwierig ist.

raschy, 2024-08-27:
Dann bin ich auch am Ende. Ich kann nur von einem anderen User berichten, der einen ähnlichen Fehler hatte, bei ihm waren IP-Adressen und Seriennummern vertauscht. Sind bei dir beide Werte richtig?

Photographix, 2024-08-28:
Hallo und guten Abend, nein bei mir ist alles richtig und korrekt. Würde er sich denn verbinden wenn die Logger Seriennummer falsch wäre? Gibt es einen Debug Modus? Das Logging steht schon auf "Alles".

raschy, 2024-08-28:
Ohne richtige Seriennummer gibt der keine Antwort. Von den kleinen WR kenne ich, dass die eine SN für den Logger und eine für den Wechselrichter selbst haben. Wurde öfters verwechselt. 
Hat deiner auch zwei Seriennummern?

Photographix, 2024-08-28:
ja hab zwei seriennummern, die vom logger ist aktuell eingetragen. die die auch in der deye cloud registriert ist / geht.

raschy, 2024-08-28:
Sollte dann richtig sein. Hast du mal einen Auszug aus deinem Log. Debug wäre der richtige Modus.

Photographix, 2024-08-28:
das log bleibt jetzt so die nächsten minuten danach 
kommt nichts mehr

ich schrieb ja extra das ich einen LAN Logger habe - ggf macht dieser was anderes. aber das war  nur eine vermutung.

`
deyeidc.0 | 2024-08-28 20:57:52.833 | debug | Connected to server
-- | -- | -- | --
deyeidc.0 | 2024-08-28 20:57:52.825 | debug | try to connect . . .
deyeidc.0 | 2024-08-28 20:57:52.824 | debug | checkUserData is ready
deyeidc.0 | 2024-08-28 20:57:52.824 | info | Retrieving data from the inverter will be done every 60 seconds
deyeidc.0 | 2024-08-28 20:57:52.824 | debug | IP address [192.168.200.65] seems to be valid.
deyeidc.0 | 2024-08-28 20:57:52.816 | info | starting. Version 0.1.0 in /opt/iobroker/node_modules/iobroker.deyeidc, node: v20.17.0, js-controller: 6.0.9
deyeidc.0 | 2024-08-28 20:57:48.088 | info | terminating
deyeidc.0 | 2024-08-28 20:57:47.565 | debug | Connection closed
deyeidc.0 | 2024-08-28 20:57:47.560 | info | Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
deyeidc.0 | 2024-08-28 20:57:47.551 | info | terminating
deyeidc.0 | 2024-08-28 20:57:47.538 | debug | [onUnload] cleaned everything up...
deyeidc.0 | 2024-08-28 20:57:47.500 | info | Got terminate signal TERMINATE_YOURSELF
`

Photographix, 2024-08-28:
[image]

Photographix, 2024-08-28:
[image]

raschy, 2024-08-28:
Mach bitte mal bei den Registerbereichen den ersten von 500 -501.

Photographix, 2024-08-29:
Guten Morgen, das bringt leider keine Veränderung. Weiterhin "connect" und alles ruhig

Photographix, 2024-08-29:
[image]

raschy, 2024-08-29:
Wie sehen denn die Spulen aus?

Photographix, 2024-08-29:
Also ich hab "nur" die json aus dem Repo reingeladen uns bis auf die IP und Seriennummer des Loggers  nichts verändert. Also auch keine Register oder Spulen angepasst.

[image]

raschy, 2024-08-29:
Die .json hat ja ein anderer User erstelllt und läuft bei ihm. Dann bin ich mit meiner Weisheit am Ende.

Photographix, 2024-08-29:
ok verstehe. Gibt es einen verbose Debugmodus des Adapters, wo ich die Kommunikation ganz konkret belauschen kann oder muss ich hier wirklich Wireshark oder ähnliches auffahren, um herauszubekommen? Was passiert? Ich wüsste auch nicht, wie ich sonst an den Wechselrichter connecten könnte in die Cloud scheint er ja zu loggen.

raschy, 2024-08-29:
Im Adapter ist nur 'debug' implementiert. Der loggt mit, was passiert. z.B: `<html>
<body>

deyeidc.0 | 2024-08-29 10:15:26.454 | debug | Response: (checkOnlineDate) {"register":0,"modbus":{"type":"Buffer","data":[1,3,2,29,10,49]}}
-- | -- | -- | --
deyeidc.0 | 2024-08-29 10:15:26.023 | debug | [computeData] ResultJson: [{"key":"DP1","value":"171.00","unit":"W","name":"DC Power PV1"},{"key":"DP2","value":"0.00","unit":"W","name":"DC Power PV2"},{"key":"DP3","value":"0.00","unit":"W","name":"DC Power PV3"},{"key":"DP4","value":"43.80","unit":"W","name":"DC Power PV4"},{"key":"DVV","value":"28.50","unit":"V","name":"Spannung Add"}]
deyeidc.0 | 2024-08-29 10:15:26.023 | debug | [computeData] {"values":"DV1+DV2","key":"DVV","name":"Spannung Add","unit":"V","factor":0}
deyeidc.0 | 2024-08-29 10:15:26.023 | debug | [computeData] {"values":"DV4*DC4","key":"DP4","name":"DC Power PV4","unit":"W","factor":0}
deyeidc.0 | 2024-08-29 10:15:26.023 | debug | [computeData] {"values":"DV3*DC3","key":"DP3","name":"DC Power PV3","unit":"W","factor":0}
deyeidc.0 | 2024-08-29 10:15:26.023 | debug | [computeData] {"values":"DV2*DC2","key":"DP2","name":"DC Power PV2","unit":"W","factor":0}
deyeidc.0 | 2024-08-29 10:15:26.023 | debug | [computeData] {"values":"DV1*DC1","key":"DP1","name":"DC Power PV1","unit":"W","factor":0}
deyeidc.0 | 2024-08-29 10:15:26.023 | debug | Data reception for 4 registersets completed
deyeidc.0 | 2024-08-29 10:15:25.993 | debug | Response: (payload) {"register":4,"modbus":{"type":"Buffer","data":[1,3,8,0,0,10,190,7,8,20,80,50]}}
deyeidc.0 | 2024-08-29 10:15:25.853 | debug | Response: (payload) {"register":3,"modbus":{"type":"Buffer","data":[1,3,16,1,29,0,60,0,0,0,0,0,0,0,0,1,36,0,15,169]}}
deyeidc.0 | 2024-08-29 10:15:25.514 | debug | Response: (payload) {"register":2,"modbus":{"type":"Buffer","data":[1,3,64,0,2,0,12,0,0,0,0,61,35,0,0,0,4,0,2,0,5,0,0,13,232,0,0,13,155,0,0,9,116,13,99,0,0,0,8,17,83,0,0,19,136,0,0,0,0,0,0,0,0,0,0,0,0,8,102,0,0,0,0,0,0,18,52,73]}}
deyeidc.0 | 2024-08-29 10:15:25.218 | debug | Response: (payload) {"register":1,"modbus":{"type":"Buffer","data":[1,3,82,0,4,1,0,2,1,50,50,49,48,48,53,48,57,50,55,0,0,0,0,18,12,7,0,1,2,1,39,18,21,0,0,78,32,0,0,4,1,0,75,0,0,0,60,24,8,29,10,11,56,0,0,0,0,10,190,7,8,20,80,18,142,0,0,0,0,0,0,19,156,0,44,0,0,0,0,0,0,0,0,0,100,25]}}
deyeidc.0 | 2024-08-29 10:13:26.410 | debug | Response: (checkOnlineDate) {"register":0,"modbus":{"type":"Buffer","data":[1,3,2,29,10,49]}}

</body>
</html>` Evtl hilft wirklich mit Wireshark zu sehen, was rausgeht und was zurück kommt. Was mir gerade in den Sinn kommt ist die Funktion 'checkOnlineDate'. Wen dein WR da nichts zurück gibt . . . ?

Photographix, 2024-08-29:
Ich habe momentan keinen Ansatz. Wireshark muss ich sehen, ob ich den am Wochenende mal aufsetzen kann. Kann aber auch sein es zieht sich wegen anderen Dingen und ich verwerfe das Projekt. Wie gesagt, in die Cloud loggt er ja scheinbar problemlos. Aber ich möchte noch einmal darauf hinweisen ich habe einen LAN Logger, keine Ahnung, was der Unterschied ist zwischen LAN und WLAN. Scheinbar scheint es da ja doch etwas zu geben. Es sieht zumindest so aus.

Photographix, 2024-08-29:
Vielleicht noch ein kleiner Fakt, zu mindestens habe ich noch getestet: mit der Seriennummer vom Wechselrichter sowie mit der Seriennummer vom LAN Logger Bekomme ich einen Connect, aber keine Daten. Kürze ich die Seriennummer einfach ein, sagt er mir die Seriennummer ist nicht korrekt und bleibt auf einem orangen Dreieck stehen.

raschy, 2024-08-29:
Das ist nur eine stumpfe Prüfung, ob die Seriennumer ungefähr richtig ist. Also kürzen bringt nichts. 
Und den Unterschied zwischen LAN und WLAN kann ich auch nicht ausmachen.

Photographix, 2024-09-09:
Was ich noch nicht gemacht haben - einen Rest vom Logger.
Ist Dir bekannt ob dann meine Cloud Daten oder andere Sachen irreparabel weg sind?

raschy, 2024-09-09:
Reset vom Logger habe ich noch nicht gemacht - war bisher auch bei keinem Nutzer nötig.
Solange die Verknüpfung zwischen Cloud und WR-Seriennummer nicht gelöscht wird, werden die Daten wohl vorhanden bleiben.
In den meisten Fällen, bei denen die Verbindung nicht funktionierte war es eine falsche Seriennummer oder das Netzwerk spinnt.

Photographix, 2024-09-15:
Ein Reset hat leider nicht geholfen.
Aber ja alle Daten bleiben erhalten. 

Im Iobroker weiterhin alle leer.

raschy, 2024-09-16:
Ich aknn mich nur wiederholen: Es liegt wohl nicht am Adapter sondern an deinem WR bzw. der Konstellation damit. Hängt an der Schnittstelle, an der du auslesen willst, noch ein anderes Gerät dran? SmartMeter oder Batterien?

Photographix, 2024-09-19:
Hallo ich bin minimal ein Stück weiter gekommen.  Ich hatte mit dem Solarman Support Kontakt die mir aber nicht helfen konnten.

Mittels Chat GPT konnte ich mir ein Powershell Script erstellen lassen was den Deye12k abfraget (bzw das Dongle) . Es war bis dahin zumindest für mich nicht klar das Funktion da ist. Das Script liefert folgende Ausgaben:

[image]

Photographix, 2024-09-19:
> Hängt an der Schnittstelle, an der du auslesen willst, noch ein anderes Gerät dran? SmartMeter oder Batterien?

Nein der Logger ist ja per LAN über Seriell angeschlossen. Über CAN Bus hängt der Akku dran.

Photographix, 2024-09-19:
wenn ich sehr lange warte - so 3-4 Min kommt nach langer Zeit sowas. 

[image]

raschy, 2024-09-19:
Auf welchem Unterbau (PC, Raspberry, usw.) läuft denn der ioBroker? Deine Auszüge aus dem Log sind immer sehr kurz. Irgendwann (nach 60 Sekunden) müsste ja mal wieder etwas kommen. Entweder "try to connect" oder "Request to register...".
PS: Kannst du mir mal das Script zur Verfügung stellen?

Photographix, 2024-09-20:
[deye.ps1.txt](https://github.com/user-attachments/files/17074117/deye.ps1.txt)

Anbei das von ChatGPT (**nicht von mir**) erstellte Script, was aus dem Dongel Werte ausliest dich aber noch codiert sind. Man muss nur die IP und die Seriennummer des Dongles ergänzen.

Ich nutze IObroker auf einem NUC in Docker mit dem Image buanet/iobroker:latest

Das Log in der View in IObroker (Protokolle) rotiert leider raus. Die Werte bleiben nicht langfristig sichtbar. Sollte da mal mehr stehen kopiere ich das gerne und hänge es hier an.

raschy, 2024-09-20:
Schaust du die Logs auf dem Handy? Also ich kann locker 50 Zeilen lesen. 
Versuch bitte mal, den Adapter direkt von Github zu installieren. Ich habe die passage um den "Frame CheckSum" mal abgeschaltet. Irgendwas stimmt mit deinem WR nicht. Evtl. Problem der Firmware.
PS: Danke fürs Script

Photographix, 2024-09-20:
Hallo nein ich schaue das ganz normal am Rechner. Vorher ein Mac mit Chrome, nun Windows mit Vivaldi.
Neuer Adapter wurde installiert. ggf etwas gesprächiger. Aber mehr als 6 Einträge kommen auch nicht. Dann rotiert der letzte raus. Ggf ein Setting in IObroker. Ich weiss nicht.

[image]

raschy, 2024-09-20:
Kannst du bitte mal alle Register und alle Coils löschen (Kannst du ja später wieder importieren) und dann nur ein Register '500 - 501' und nur die Coil '500' eintragen. Da muss was völlig Quer hängen.

Photographix, 2024-09-21:
[image]

[image]

[image]

raschy, 2024-09-21:
Danke für die Mühe. So wie das aussieht, kommt keine Antwort vom WR. Eigentlich müsste nach deiner '
silly | Request to register set 1 > a5170010450b00db80eef70200000000000000000000000000000103003c000144064415' nach weniger als 100ms ein 
'debug | Response: (payload) {"register":1,"modbus":{"type":"Buffer","data":[1,3,2,0,10,56]}}' (oder so ähnlich) kommen. Bei dir kommt aber erst mal nix und dann ein 'warn undefined' - woher auch immer.

Photographix, 2024-09-22:
In einem Chat habe ich die Modbus Spezifikation von Deye gefunden (angeblich) ggf hilft das uns hier weiter? Ich hab diese dir mal per Mail an deine Github Adresse geschickt.

Photographix, 2024-10-28:
Ich bin in dieser Fehlersuche leider nicht mehr involviert. Ich bin zu Home Assistant gewechselt (auch wegen einiger anderen iobroker Probleme) und habe mit dem Deye12k an Home Assistant keine Probleme.

Ticket kann ggf. geschlossen werden.

raschy, 2024-10-28:
so sei es.

closed 2024-10-28 by raschy