raschy/ioBroker.deyeidc #124: Adapter läßt sich nicht installieren IOBroker

MarkX81, 2024-08-23:
Hallo 

Ich hatte schon mehrmals probiert den Adapter zu installieren, aber ich erhalte immer Error Code 25

Hier mal ein LOG: `
$ iobroker add deyeidc --host Raspi4

NPM version: 10.8.2

Installing iobroker.deyeidc@0.1.0... (System call)

npm error code 1npm error path /opt/iobroker/node_modules/bufferutilnpm error command failednpm error command sh -c node-gyp-buildnpm error make: Verzeichnis �/opt/iobroker/node_modules/bufferutil/build� wird betretennpm error CC(target) Release/obj.target/bufferutil/src/bufferutil.onpm error make: Verzeichnis �/opt/iobroker/node_modules/bufferutil/build� wird verlassennpm error make: cc: Datei oder Verzeichnis nicht gefundennpm error make: *** [bufferutil.target.mk:106: Release/obj.target/bufferutil/src/bufferutil.o] Fehler 127npm error gyp ERR! build error npm error gyp ERR! stack Error: `make` failed with exit code: 2npm error gyp ERR! stack at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:209:23)npm error gyp ERR! System Linux 6.1.0-rpi8-rpi-v8npm error gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"npm error gyp ERR! cwd /opt/iobroker/node_modules/bufferutilnpm error gyp ERR! node -v v20.17.0npm error gyp ERR! node-gyp -v v10.1.0npm error gyp ERR! not oknpm error A complete log of this run can be found in: /home/iobroker/.npm/_logs/2024-08-23T08_48_13_259Z-debug-0.log

host.Raspi4 Cannot install iobroker.deyeidc@0.1.0: 1

ERROR: Process exited with code 25`

Hat jemand eine Idee woran dies liegen könnte. 

Vielen Dank

raschy, 2024-08-23:
Siehe hier: Fehler bei der Installation CODE25 [#81](https://github.com/raschy/ioBroker.deyeidc/issues/81)

MarkX81, 2024-08-25:
Danke, aber ich verstehe noch nicht den Lösungsweg. 
Bufferutil hab ich installiert, aber der Adapter läßt sich trotzdem nicht installieren.

raschy, 2024-08-25:
Dann schau mal hier, ob da was für dich dabei ist: 
https://forum.iobroker.net/topic/54338/error-process-exited-with-code-25
https://forum.iobroker.net/topic/53619/error-25-beim-installieren-von-adaptern

MarkX81, 2024-08-25:
Was ich halt komisch finde ist, dass ich jeden beliebigen Adapter installieren kann. 
Nur der Deye Adapter läßt sich nicht installieren. 
Versuche mal, ob deine Tipps helfen. Vielen Dank schon mal dafür.

raschy, 2024-08-25:
Ich bin sicher, dass es nicht an dem Adapter liegt. Es gab schon mal User, bei denen waren noch "Restbestände" einer vorherigen (missglückten) Installation zurückgeblieben.

MarkX81, 2024-08-26:
Kann ich evtl. irgendein Log posten, damit man die "Restbestände" sieht?

raschy, 2024-08-26:
Du kannst mal die Versionsstände von nodejs, npm, js-controller und admin angeben. Vielleicht sieht man dann schon etwas. 
Die 'Restbestände' sind wahrscheinlich unter /opt/iobroker/node_modules zu sehen.

MarkX81, 2024-08-27:
Node.js: v20.17.0
NPM: 10.8.2
JS-Controller: 6.0.11
admin ?? - Was meinst du damit. 
In dem genannten Ordner befinden sich 660 Unterordner.

raschy, 2024-08-27:
Na, der admin hat doch auch eine Versionsnummer. Aber es scheint, du bist auf dem aktuellen Stand.
Schreib dein Problem doch mal ins ioBroker-Forum. Da gibt es Spezialisten für soetwas.

closed 2024-10-01 by raschy