raschy/ioBroker.deyeidc #156: Version 0.1.3 [checkConnected] Error: connect EHOSTUNREACH

MarkSau, 2025-01-05:
Seit dem Update flutten meine beiden Instanzen den Log

2025-01-05 17:07:59.677 - error: deyeidc.0 (978) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.75:8899
2025-01-05 17:08:02.749 - error: deyeidc.1 (1012) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.76:8899
2025-01-05 17:08:59.677 - error: deyeidc.0 (978) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.75:8899
2025-01-05 17:09:02.749 - error: deyeidc.1 (1012) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.76:8899
2025-01-05 17:09:59.677 - error: deyeidc.0 (978) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.75:8899
2025-01-05 17:10:02.749 - error: deyeidc.1 (1012) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.76:8899
2025-01-05 17:10:59.677 - error: deyeidc.0 (978) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.75:8899
2025-01-05 17:11:02.749 - error: deyeidc.1 (1012) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.76:8899
2025-01-05 17:11:59.678 - error: deyeidc.0 (978) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.75:8899
2025-01-05 17:12:02.749 - error: deyeidc.1 (1012) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.76:8899
2025-01-05 17:12:59.677 - error: deyeidc.0 (978) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.75:8899
2025-01-05 17:13:02.749 - error: deyeidc.1 (1012) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.76:8899
2025-01-05 17:13:59.677 - error: deyeidc.0 (978) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.75:8899
2025-01-05 17:14:02.749 - error: deyeidc.1 (1012) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.76:8899
2025-01-05 17:14:59.677 - error: deyeidc.0 (978) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.75:8899
2025-01-05 17:15:02.749 - error: deyeidc.1 (1012) [checkConnected] Error: connect EHOSTUNREACH 192.168.1.76:8899

Das passiert ab dem Moment, an dem die WRs (Deye 600 & 800), wegen Mangels an Sonne, abschalten und endet am nächsten Tag, wenn wieder Sonne da ist und die WRs wieder "online" gehen.

Ich habe erstmal wieder die V0.1.2 eingespielt. Damit kommen die Meldungen nicht mehr.

Ilovegym66, 2025-01-05:
war hier auch so, auch zurueck auf 0.1.2 
Node 20, Admin 7.4.10, Js-controller 7.0.6

raschy, 2025-01-05:
Gut dass ich eben selbst mal die 0.1.3 installiert habe. Da bin ich wohl übers Ziel etwas hinausgeschossen. Versuch ich morgen zu beheben.

raschy, 2025-01-11:
Fehler behoben in Version 0.1.4
Bug fixed in version 0.1.4