raschy/ioBroker.deyeidc #157: Wierd spikes in data

bjoern-le, 2025-01-12:
I have a SUN 12k. Usually the data read by the adapter looks good an the P_GRID_CT_TOTAL (currently 43W) and P_GRID_LD_TOTAL (currently 407) are very close to the 427 W shown by the smart meter. 

But: When there is enough current generated by the PV so that I don't need energy from the grid the consumption goes to values like 65497 as soon as I export to the grid. I think this is related to the data type (likely an unsigned int), but it is strange to see the spikes in the graphs. Is there a way to change this behaviour and show negative numbers?

Steps to reproduce the behavior:
1. Export to grid
2. Look in history for P_GRID_CT_TOTAL  -> Values > 60000W instead of negative W

**Expected behavior**  
Negative Watts when energy gets exported.

**Versions:**  
 - Adapter version: 0.1.1
 - JS-Controller version: 6.0.11
 - Node version: v20.17.0
 - Operating system: docker container inside raspbian

bjoern-le, 2025-01-16:
I am very sorry for the delay. The coils and so on should be the ones from the sun12k config file. 

The config for coil P_GRID_CT_TOTAL is

`{
      "register": 619,
      "rules": 1,
      "key": "P_GRID_CT_TOTAL",
      "name": "P GRID CT TOTAL",
      "unit": "W",
      "factor": 0,
      "nullable": false
    },`

raschy, 2025-01-16:
Please try changing the rules from ‘1’ to '3'

bjoern-le, 2025-01-22:
Quick update: The rules are updated and I am waiting for the fog to disappear so I can generate enough power to export it.

bjoern-le, 2025-01-24:
Today there is enough sunlight. Unfortunately, the values still show 65xxx Watt if I send power to the grid for the Value P_GRID_LD_TOTAL with rule set to 3.

ardu1, 2025-02-22:
Hello, I had the same problem, also with all P_GRID_ coils. After changing the rules to 2 it seems correct. 

Versions:

Adapter version: 0.2.0
JS-Controller version: 7.0.6
Node version: v20.18.03
Operating system: Linux 12 bookworm

raschy, 2025-02-22:
I have taken note of this change in the configuration and will incorporate it into the files during the next edit. Thank you.

closed 2025-02-22 by raschy

ardu1, 2025-02-22:
:-)