raschy/ioBroker.deyeidc #159: Update stable version in repo from 0.1.1 to 0.1.4

GermanBluefox, 2025-01-27:
# Think about update stable version to 0.1.4
**Version**: stable=**0.1.1** (106 days old) => latest=**0.1.4** (15 days old)
**Installs**: stable=**890** (63.17%), latest=**76** (5.39%), total=**1409**

Click to use [developer portal](https://www.iobroker.dev/adapter/deyeidc/releases)
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L438)

**Do not close this issue manually as a new issue will be created if condition for update still exists.**

Please drop a comment if any reason exists which blocks updating to version 0.1.4 at this time.

Note: This is an automatically generated message and not personally authored by bluefox!
      @mcm1957 for evidence

raschy, 2025-01-27:
The version 0.1.4 still contains the @iobroker/adapter-core: 3.2.2, do I have to make a new version for the newer core?

mcm1957, 2025-01-27:
core 3.2.2 is ok for release to stable.
3.2.3 is only recommended not required. 
Update with next regular release is sufficient.

raschy, 2025-02-06:
Version 0.1.4 is oboslet.  
A new version 0.2.0 with the same range of functions was created because there were massive problems with the upload.

closed 2025-02-06 by raschy