raschy/ioBroker.deyeidc #181: Daylight Saving Time Issue: Incorrect 'Production Today' Values Logged with ioBroker and InfluxDB2

mlml37, 2025-04-02:
**Describe the bug**  
I have an M80G4-EU and set it up with the deyeidc adapter according to the instructions in ioBroker. The values ​​are stored in an influxdb2 database.

Since the switch to daylight saving time, the adapter logs several data points for Etdy_ge0 - "Production Today" with the final value of the previous day after the inverter starts. The first value logged after startup is 0, followed by three incorrect values, and then the normal values ​​are logged again. The incorrect measured values ​​do not appear in the cloud.

**To Reproduce**  
Steps to reproduce the behavior:
The inverter polling rate is set to 60 seconds in the adapter.

**Expected behavior**  
No recording of false measurements.

**Screenshots & Logfiles**  

[image]

[image]

**Versions:**  
 - Adapter version: 0.2.0
 - JS-Controller version: 7.0.6
 - Node version: 20.19.0
 - Operating system: docker (official image - v10.0.0)

**Additional context**  
Add any other context about the problem here.

raschy, 2025-04-02:
The adapter only reads out what the inverter supplies. 
Please also observe in the following days what data is supplied when the inverter is restarted.

mlml37, 2025-04-04:
It was the inverter. After two days of hiccups, it's now delivering error-free readings again.

closed 2025-04-04 by mlml37