raschy/ioBroker.deyeidc #182: deyeidc

avtomat78, 2025-04-05:
The adapter has stopped reading the network inverter since version 0.1.1
SUN-8K-G

raschy, 2025-04-05:
Please try Release 0.2.0

avtomat78, 2025-04-05:
0.2.0 doesn't work either! the latest working version for my inverter is 0.1.0

raschy, 2025-04-05:
Could I then also have a few lines from the log, preferably by setting the adapter to 'debug' beforehand.

avtomat78, 2025-04-05:
archivo adjunto

[does not work.txt](https://github.com/user-attachments/files/19615614/does.not.work.txt)
[works.txt](https://github.com/user-attachments/files/19615615/works.txt)

raschy, 2025-04-05:
here is the reason: node: v18.20.8,
Please update your system to node v20.x.x

avtomat78, 2025-04-05:
it didn't help, it doesn't even create a folder with a serial number!

[does not work.txt](https://github.com/user-attachments/files/19616090/does.not.work.txt)

raschy, 2025-04-05:
Browser refresh? It actually looks good: deyeidc.0 (2559) Data reception for 4 registersets completed.
Perhaps the poll rate of 10 seconds is a bit high.

avtomat78, 2025-04-05:
I see it for 60 seconds, it doesn't help, the folder with the serial number is not created, it works on version 0.1.0!
I also have a hybrid deye, it works with version 0.2.0, but this one doesn't want to

raschy, 2025-04-05:
The next attempt would be to set the instance to log level "all" (=silly)  and store a log file here again.

avtomat78, 2025-04-06:
switched to all

[does not work.txt](https://github.com/user-attachments/files/19619669/does.not.work.txt)

raschy, 2025-04-06:
I cannot reproduce this error. If I delete the folder 0.xxxx, the folder is completely recreated the next time valid data is received.

raschy, 2025-04-06:
Have you ever tried uninstalling and then reinstalling? Or do you happen to have an ioBroker test installation to hand?

raschy, 2025-04-06:
Then I can't think of anything more to say here at the moment. Sorry.

avtomat78, 2025-04-06:
It's a pity, we'll have to use the old version.

raschy, 2025-04-06:
Yes, sure, that's a possibility. However, I would still like to know the reason why the folder is not being created.

avtomat78, 2025-04-06:
It will be good if you can fix it!