raschy/ioBroker.deyeidc #183: Two small issues in sample config for 12kW

peteruhrig, 2025-04-18:
Dear all,

Thanks for the great work! This now allows me to get my Deye data out without going via their cloud solution. I have two small issues with the config for the 12kW device:

In the file `deyeidc.Sun12KP3.json`, the last two registers are odd:

```
		{
			"register": 592,
			"rules": 1,
			"key": "Battery_Current_Corrected_AH",
			"name": "Battery Current",
			"unit": "AH",
			"factor": 0,
			"nullable": false
		},
		{
			"register": 590,
			"rules": 1,
			"key": "Battery_Output_Power",
			"name": "Battery Power",
			"unit": "W",
			"factor": 2,
			"nullable": false
		}
```
The first one seems to give us some battery current, but it is in a unit called "AH" instead of "A". Or should it be the battery capacity, which for consistency's sake we probably should have in "Ah". Either way, for my setup this gives me 200, which I cannot understand, because I have a 300AH battery setup (3x 5.12 kWh) and I have currently limited charging and discharging to 100A on the inverter. As it says "Corrected", what exactly does it correct for?

In the second case, at least for me, the factor needs to be 0, not 2 (might be a copy-paste issue from the battery voltage further up).

My inverter is a SUN-12K-SG04LP3-EU, my batteries are Pytes E-BOX-48100R.

All the best,
Peter

peteruhrig, 2025-04-19:
Now there is sun, there is another issue with the second register. The value of "rule" needs to be set to 2, otherwise it will report 65 kW values here, similar to issue #157.

raschy, 2025-04-19:
As I have often written, I am dependent on feedback from users, as I do not have a hybrid inverter.
Therefore, after successful verification of the data obtained, please feel free to post the correct settings here.