raschy/ioBroker.deyeidc #200: Pulldown Menue with options

bobybobybob, 2025-07-12:
I have the program running and see the total watts. Other data such as PV1/2 volt, PV1/2 watt and others not. It refers to using the data for registers from some JSON files. Why no pull down menu where you can select some of them as default. It is very time consuming to type all this manually without knowing whether they are the right ones. An autodetect would be great

raschy, 2025-07-13:
This is a universal adapter that the user can customize depending on which system is available and which data he wants to see. How am I supposed to recognize a standard?

bobybobybob, 2025-07-13:
it is simply frustrating to enter all the individual values manually. It did work but was no fun.  But you're probably right if you want it to work for several different submodels.

closed 2025-07-13 by bobybobybob